RobotLocomotion/drake #2541: RigidBodyTreeURDF doesn't correctly parse joint torque limits

manuelli, 2016-06-10:
When parsing the Atlas urdf (e.g. examples/Atlas/urdf/atlas_minimal_contact.urdf) the RigidBodyTreeURDF class looks for effort limits in the <transmission> tab. However the actual effort limits are part of the <joint> tab. This causes effort limits to be set to [-inf,inf]. We have repeatedly commanded outside of effort limits on Valkyrie due to this issue.

liangfok, 2016-06-10:
I can take a look at this soon.

liangfok, 2016-06-10:
For reference [here](http://wiki.ros.org/urdf/XML/joint) is the specification for a joint in URDF. The torque limit is attribute "effort" within tag `<limit>`.

liangfok, 2016-06-11:
Unfortunately, `RigidBodyTree.cpp` does not even search for effort limits:

https://github.com/RobotLocomotion/drake/blob/master/drake/systems/plants/RigidBodyTreeURDF.cpp#L337

manuelli, 2016-06-11:
It does, but it's in the transmission object of the urdf. https://github.com/RobotLocomotion/drake/blob/master/drake/systems/plants/RigidBodyTreeURDF.cpp#L530

liangfok, 2016-06-11:
Never mind, the limit is parsed elsewhere in `RigidBodyTreeURDF.cpp`:

https://github.com/RobotLocomotion/drake/blob/master/drake/systems/plants/RigidBodyTreeURDF.cpp#L525

liangfok, 2016-06-11:
The problem lies on line: https://github.com/RobotLocomotion/drake/blob/master/drake/systems/plants/RigidBodyTreeURDF.cpp#L525

In this case `joint_node` is the XML node _within_ a `transmission` element, _not_ within an actual joint element. I believe the solution is to parse the joint effort limits when parsing the joint's XML element, and then refer to these values when parsing the transmission element.

closed 2016-06-14 by sherm1