RobotLocomotion/drake #2549: Document examples used for optimization tests

naveenoid, 2016-06-13:
The sample problems used in optimization tests need to be described in the documentation.

Lucy-tri, 2016-06-13:
The optimization tests often reference a book of test, but that is a paper book and so is not always available to look at. Would be nice to document the equations and other details in the code.

naveenoid, 2016-06-14:
@ggould-tri @sammy-tri @siyuan-tri This is the issue I referenced. Perhaps in documenting the example systems it would make sense to 'Doxygen-fy' it?

ggould-tri, 2016-06-14:
IMO (and @sammy-tri will probably disagree here) if the book is only available as a paper artifact, we probably need to copy the problem statement into the file -- we are a distributed organization, and I don't relish the opportunity to create a corporate library to manage our trove of tomes.  If we can get the book in electronic, permalink-able form, a permalink to the problem or at least the relevant section should suffice.

naveenoid, 2016-06-14:
Also consider that @siyuanfeng-tri will be adding in some robot tests and the equations of those problems should ideally exist in the comments along with these problems.

ggould-tri, 2016-06-14:
Might it make sense to link to the GAMS files for the problems, eg https://titan.princeton.edu/TestProblems/chapter10/ex10.1.5.gms ?  Those should be more complete and correct than any comments we might write.

jwnimmer-tri, 2019-03-04:
Kicking this over to @hongkai-dai for a decision / resolution.  I don't think @naveenoid is working on it.

hongkai-dai, 2019-03-04:
I will close this issue, as we already have a lot of examples, and I added a bunch of documentation for each of the example.

closed 2019-03-04 by hongkai-dai