RobotLocomotion/drake #2550: Should Cars Example Subclass RigidBodySystem?

liangfok, 2016-06-13:
This technique is used in #2542 and in SpringMassSystem. Maybe it's useful for the migration to System 2.0?

jwnimmer-tri, 2016-06-13:
(1) Public inheritance for implementation re-use is an anti-pattern.  (Prefer has-a to is-a.)  For example, I am not sure why I would want `CarsSystem` to allows its callers to `addForceElement` into it?

(2) I don't think `SpringMasSystem` is extending `RigidBodySystem`, or anything even analogous to it.

liangfok, 2016-06-13:
`SpringMassSystem` extends `sytsems::ContinuousSystem`, which I assume is System 2.0's version of `RigidBodySystem`. See: 

https://github.com/RobotLocomotion/drake/blob/master/drake/examples/spring_mass/spring_mass_system.h#L71

@amcastro-tri, can you comment?

david-german-tri, 2016-06-13:
I agree with @jwnimmer-tri on both counts, although that's not necessarily a criticism of #2542, which I haven't studied yet.

`ContinuousSystem` is definitely not analogous to `RigidBodySystem`.  It's analogous to the "System concept" in System1.

jwnimmer-tri, 2016-06-13:
Right.  `ContinuousSystem` is an abstract base class, not a dynamics implementation.

amcastro-tri, 2016-06-13:
IMO:

1. If CarSystem "is a" RigidBodySystem, the it does make sense to add a force element. Eg: to simulate wind force.
2. SpringMassSystem implements the ContinuousySystem concept, does not extend it. The CarSystem might like to implement the RigidBodySystem concept and inheriting from it makes sense to me. It's a more concrete type of a RigidBodySystem.

jwnimmer-tri, 2016-06-13:
But the RBS is the world, not a single System.  I don't really want my CarSystem to extend the world?

liangfok, 2016-06-13:
Aha: https://google.github.io/styleguide/cppguide.html#Inheritance

I'm now leaning towards concluding that the answer to this issue is "no".

amcastro-tri, 2016-06-13:
@jwnimmer-tri: that is probably our basic underlying problem, that RBS is our world. 
BTW: I don't think we should spend time making our `CarSystem` inheriting from System 1.0. However I can see how this will be a useful concept when System 2.0 arrives. That plus a "world" concept that will allows us to place (instantiate) several CarSystem's (thoroughly unit tested) in the world.

liangfok, 2016-06-13:
> That plus a "world" concept that will allows us to place (instantiate) several CarSystem's (thoroughly unit tested) in the world.

See: https://github.com/RobotLocomotion/drake/issues/2318

liangfok, 2016-06-13:
I believe we've converged to agreeing that as long as there is only _one_ `RigidBodySystem` representing the whole world, an inheritance model makes less sense relative to composition. However, if and when Drake is updated to support multiple `RigidBodySystem` objects all within a separate `World`, an inheritance model may make more sense. Since updating Drake to support multiple `RigidBodySystem` objects is still under discussion (see: https://github.com/RobotLocomotion/drake/issues/2318 and https://github.com/RobotLocomotion/drake/pull/2426#issuecomment-224260420), I'm going to close this thread for now.

closed 2016-06-13 by liangfok