RobotLocomotion/drake #2554: Tests fail when SNOPT is enabled on Windows

jamiesnape, 2016-06-13:
Blocks #2352.

**32-bit**

* testOptimizationProblem - OTHER_FAULT
* testIKpointwise - OTHER_FAULT
* testIKMoreConstraints - OTHER_FAULT
* testIK - OTHER_FAULT
* comparePriusURDFtoSDF - Failed

See https://drake-cdash.csail.mit.edu/viewTest.php?onlyfailed&buildid=34200

**64-bit**

* testOptimizationProblem - Failed

See https://drake-cdash.csail.mit.edu/viewTest.php?onlyfailed&buildid=34195

david-german-tri, 2016-07-20:
@ggould This is blocking SNOPT-on-Windows CI.  Do you care?  If not, it may make sense to dial down the priority.

ggould-tri, 2016-08-03:
I believe I have now successfully reproduced the problem.
The error is solver unavailable; visual studio debugger strongly suggests that NoSnopt is being linked in rather than SnoptSolver.  Investigation continuing but it's definitely build rule fail of some sort.

ggould-tri, 2016-08-03:
As near as I can tell `pods_find_pkg_config(snopt_c)` is failing or inconsistent on windows when snopt is present and built.

In particular the `plants/test` CMakeLists says to only add `testIK` when `snopt_c_FOUND` is true.  And the solvers CMakeLists says to only add NoSnopt when `snopt_c_FOUND` is false.  Nonetheless `testIK` sees `snopt_solver_->available()` == false, which is only provided by NoSnopt.

I have now spent two days chasing this issue.  It is a slog through windows and cmake nonsense to no clear purpose.  I am downprioritizing it; windows can burn.

jamiesnape, 2016-10-05:
Closing since Windows is no longer supported.

closed 2016-10-05 by jamiesnape