RobotLocomotion/drake #2565: The ubuntu.rst instructions for download-all do not work

jwnimmer-tri, 2016-06-15:
I suspect that #2487 has broken the documented Ubuntu workflow.

On Ubuntu 14.04 Trusty, following `from_source.rst` and then `ubuntu.rst` from a new from-scratch build gives me:

```
jwnimmer@call:~$ git clone https://github.com/RobotLocomotion/drake.git drake-distro
Cloning into 'drake-distro'...
remote: Counting objects: 93229, done.
remote: Compressing objects: 100% (103/103), done.
remote: Total 93229 (delta 69), reused 8 (delta 8), pack-reused 93118
Receiving objects: 100% (93229/93229), 186.78 MiB | 3.13 MiB/s, done.
Resolving deltas: 100% (71784/71784), done.
Checking connectivity... done.

jwnimmer@call:~$ cd drake-distro/
jwnimmer@call:~/drake-distro$ env CC=gcc-4.9 CXX=g++-4.9 make options
... (key in: c c g) ...
Configuring with CMAKE_FLAGS: -DCMAKE_BUILD_TYPE=Release -DPOD_BUILD=TRUE
-- The C compiler identification is GNU 4.9.3
-- The CXX compiler identification is GNU 4.9.3
-- Check for working C compiler: /usr/bin/gcc-4.9
-- Check for working C compiler: /usr/bin/gcc-4.9 -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/g++-4.9
-- Check for working CXX compiler: /usr/bin/g++-4.9 -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- CMAKE_INSTALL_PREFIX=/home/jwnimmer/drake-distro/build
-- Found Git: /usr/bin/git (found version "1.9.1") 
-- Found Java: /usr/bin/java (found suitable version "1.8.0.91", minimum required is "1.6") 
-- Looked for MATLAB but could not find it.
-- Preparing to build cmake with dependencies: 
Submodule 'drake/cmake' (https://github.com/RobotLocomotion/cmake.git) registered for path 'drake/cmake'
-- Preparing to build eigen with dependencies: 
Submodule 'externals/eigen' (https://github.com/RobotLocomotion/eigen-pod.git) registered for path 'externals/eigen'
-- Preparing to build googletest with dependencies: 
Submodule 'externals/googletest' (https://github.com/google/googletest.git) registered for path 'externals/googletest'
-- Preparing to build lcm with dependencies: 
Submodule 'externals/lcm' (https://github.com/RobotLocomotion/lcm-pod.git) registered for path 'externals/lcm'
-- Preparing to build bot_core_lcmtypes with dependencies: lcm
Submodule 'externals/bot_core_lcmtypes' (https://github.com/jhoare/bot_core_lcmtypes) registered for path 'externals/bot_core_lcmtypes'
-- Preparing to build libbot with dependencies: lcm
Submodule 'externals/libbot' (https://github.com/RobotLocomotion/libbot.git) registered for path 'externals/libbot'
-- Preparing to build bullet with dependencies: 
Submodule 'externals/bullet' (https://github.com/RobotLocomotion/bullet-pod.git) registered for path 'externals/bullet'
-- Preparing to build director with dependencies: bot_core_lcmtypes;lcm;libbot
Submodule 'externals/director' (https://github.com/RobotLocomotion/director.git) registered for path 'externals/director'
-- Preparing to build swigmake with dependencies: 
Submodule 'externals/swigmake' (https://github.com/rdeits/swigmake.git) registered for path 'externals/swigmake'
-- Preparing to build swig_matlab with dependencies: 
Submodule 'externals/swig_matlab' (https://github.com/rdeits/swig-matlab-pod.git) registered for path 'externals/swig_matlab'
-- Preparing to build yaml_cpp with dependencies: 
Submodule 'externals/yaml_cpp' (https://github.com/jbeder/yaml-cpp.git) registered for path 'externals/yaml_cpp'
-- Preparing to build ipopt with dependencies: 
Submodule 'externals/ipopt' (https://github.com/RobotLocomotion/ipopt-mirror.git) registered for path 'externals/ipopt'
-- Preparing to build nlopt with dependencies: 
Submodule 'externals/nlopt' (https://github.com/stevengj/nlopt.git) registered for path 'externals/nlopt'
-- Preparing to build drake with dependencies: bot_core_lcmtypes;bullet;cmake;eigen;googletest;lcm;libbot;nlopt;ipopt;swig_matlab;swigmake;yaml_cpp
-- Preparing to install precompiled snopt
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jwnimmer/drake-distro/pod-build

jwnimmer@call:~/drake-distro$ make download-all
Configuring with CMAKE_FLAGS: -DCMAKE_BUILD_TYPE=Release -DPOD_BUILD=TRUE
-- CMAKE_INSTALL_PREFIX=/home/jwnimmer/drake-distro/build
-- Looked for MATLAB but could not find it.
-- Preparing to build cmake with dependencies: 
-- Preparing to build eigen with dependencies: 
-- Preparing to build googletest with dependencies: 
-- Preparing to build lcm with dependencies: 
-- Preparing to build bot_core_lcmtypes with dependencies: lcm
-- Preparing to build libbot with dependencies: lcm
-- Preparing to build bullet with dependencies: 
-- Preparing to build director with dependencies: bot_core_lcmtypes;lcm;libbot
-- Preparing to build swigmake with dependencies: 
-- Preparing to build swig_matlab with dependencies: 
-- Preparing to build yaml_cpp with dependencies: 
-- Preparing to build ipopt with dependencies: 
-- Preparing to build nlopt with dependencies: 
-- Preparing to build drake with dependencies: bot_core_lcmtypes;bullet;cmake;eigen;googletest;lcm;libbot;nlopt;ipopt;swig_matlab;swigmake;yaml_cpp
-- Preparing to install precompiled snopt
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jwnimmer/drake-distro/pod-build
Scanning dependencies of target swigmake-update
[  0%] Creating directories for 'swigmake'
[  0%] Performing download step for 'swigmake'
fatal: destination path 'externals/swigmake' already exists and is not an empty directory.
Clone of 'https://github.com/rdeits/swigmake.git' into submodule path 'externals/swigmake' failed
make[4]: *** [swigmake-prefix/src/swigmake-stamp/swigmake-download] Error 1
make[3]: *** [CMakeFiles/swigmake-update.dir/all] Error 2
make[2]: *** [CMakeFiles/download-all.dir/rule] Error 2
make[1]: *** [download-all] Error 2
make: *** [download-all] Error 2
```

Did I make a mistake anywhere?

jwnimmer-tri, 2016-06-16:
The work-around appears to be to build using `ninja` per #2555.

jwnimmer-tri, 2016-06-16:
Assuming that this a true problem, and not an error on my part... besides first fixing the immediate defect, the second question is then how jenkins let this defect escape detection.

liangfok, 2016-06-16:
I am seeing this error too.

bradking, 2016-06-16:
> second question is then how jenkins let this defect escape detection.

The pod-build mode is not tested by the CI system.  See #2484.

PR #2567 (2016-06-16): Fix submodule download in pod-build mode
When using the PODS Makefile for an in-source build, the `POD_BUILD`
option to CMake causes `pod-build` directories to be created in the
externals' source trees.  The `pod-build` directory prevents the
`git clone` performed by `git submodule update` during the download
step from seeing an empty directory, so it refuses to clone.  Teach
the download step in `POD_BUILD` mode to remove the `pod-build`
directory prior to cloning.

Fixes #2565.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2567)
commit 800ab56c (2016-06-16): Fix submodule download in pod-build mode

When using the PODS Makefile for an in-source build, the `POD_BUILD`
option to CMake causes `pod-build` directories to be created in the
externals' source trees.  The `pod-build` directory prevents the
`git clone` performed by `git submodule update` during the download
step from seeing an empty directory, so it refuses to clone.  Teach
the download step in `POD_BUILD` mode to remove the `pod-build`
directory prior to cloning and restore it afterward.

Fixes #2565.
closed 2016-06-16 by david-german-tri