RobotLocomotion/drake #2566: Unable to Compile Drive in RelWithDebugInfo Mode Using Ninja

liangfok, 2016-06-16:
Using a PC with 16 GB of RAM, I am unable to compile Drake using `ninja` in `RelWithDebugInfo` mode even when limiting the number of processes to 1:
```
$ cmake ../drake -G Ninja -DCMAKE_BUILD_TYPE:STRING=RelWithDebInfo -DDISABLE_MATLAB=ON
$ ninja -j1
...
FAILED: examples/Pendulum/CMakeFiles/runPendulumLQR.dir/runLQR.cpp.o 
/usr/lib/ccache/g++-4.9    -Igenerated -I/home/liang/dev/drake/drake/.. -Iexports -isystem /home/liang/dev/drake-build/install/include -Ilcmgen -isystem /usr/include/glib-2.0 -isystem /usr/lib/x86_64-linux-gnu/glib-2.0/include -isystem /home/liang/dev/drake-build/install/include/eigen3 -I/home/liang/dev/drake/drake/thirdParty/spruce/include -I/home/liang/dev/drake/drake/thirdParty/cimg -Werror=all -DGTEST_DONT_DEFINE_FAIL=1 -DGTEST_DONT_DEFINE_SUCCEED=1 -DGTEST_DONT_DEFINE_TEST=1 -Wno-sign-compare -O2 -g -DNDEBUG -fPIE   -std=gnu++11 -MMD -MT examples/Pendulum/CMakeFiles/runPendulumLQR.dir/runLQR.cpp.o -MF examples/Pendulum/CMakeFiles/runPendulumLQR.dir/runLQR.cpp.o.d -o examples/Pendulum/CMakeFiles/runPendulumLQR.dir/runLQR.cpp.o -c /home/liang/dev/drake/drake/examples/Pendulum/runLQR.cpp
virtual memory exhausted: Cannot allocate memory
```

I am, however, able to compile in `Release` mode:

```
$ cmake ../drake -G Ninja -DCMAKE_BUILD_TYPE:STRING=Release -DDISABLE_MATLAB=ON
$ ninja -j1
...
[146/146] Completed 'drake'
```

The problem described above appears to have arose after https://github.com/RobotLocomotion/drake/pull/2487 was merged.

jwnimmer-tri, 2016-06-16:
Very likely a duplicate of #2074.

david-german-tri, 2016-06-16:
I agree this is likely related to #2074, but @liangfok, `virtual memory exhausted` makes me wonder if your swap partition is unreasonably small.

liangfok, 2016-06-16:
It looks like my swap size is about 1.5GB:

```
$ free -h
             total       used       free     shared    buffers     cached
Mem:           15G       5.3G        10G       341M       342M       2.1G
-/+ buffers/cache:       2.9G        12G
Swap:         1.4G       262M       1.1G
```

Is that unreasonably now?

david-german-tri, 2016-06-16:
Terrifyingly, IMO.  Mine is 31GB.

liangfok, 2016-06-16:
OK I will up my swap size. Thanks.

david-german-tri, 2016-06-17:
@liangfok Do you feel we can close this as a duplicate of #2074?

liangfok, 2016-06-17:
Sure, to decrease your cognitive load, I will assume this is a duplicate of #2074. I will reopen it if I continue to run into a memory issue after #2074 is resolved.

closed 2016-06-17 by liangfok