RobotLocomotion/drake #2568: Documentation for Cars Examples Must be Updated to Support Out Of Source Builds

liangfok, 2016-06-16:
In anticipation of https://github.com/RobotLocomotion/drake/pull/2555 landing, the documentation for the cars examples (and probably other examples that rely on model files) need to be updated since the models remain in-source.

For example, the command to launch the car simulation using Drake + ROS is now:
```
$ cd [drake-build]
$ ./drake/bin/car_sim_lcm_and_ros [drake-distro]/drake/examples/Cars/models/prius/prius_with_lidar.sdf [drake-distro]/drake/examples/Cars/models/stata_garage_p1.sdf
```

jwnimmer-tri, 2016-06-16:
If pod-build makefile support is being removed, then it's a pre-merge review defect on #2555 that the examples documentation is broken.  If the makefile support is intact, I don't see why this issue needs attention.

liangfok, 2016-06-16:
#2555 sttes that `ninja` is a "supported" way to build Drake. Since it is "supported", we should update the example documentation to reflect this style of build as well.

david-german-tri, 2016-06-16:
I think a good goal would be to make all the documentation paths relative to a `build` and `distro`, (where `build` in the PODs case is `pod-build`) and I agree with both of you that `priority: low` is appropriate.

liangfok, 2016-06-16:
Agreed. I did something like that here with "`[build artifacts directory]`": http://drake.mit.edu/unit_testing_instructions.html#defining-the-build-artifacts-directory

jwnimmer-tri, 2017-03-11:
@liangfok Is this still relevant now that only Bazel is documented?

liangfok, 2017-03-11:
I think think the Bazel-based documentation is sufficient and shall thus close this issue.

closed 2017-03-11 by liangfok