RobotLocomotion/drake #2578: IPOPT causes linker errors when enabled on Mac with MATLAB

sammy-tri, 2016-06-17:
See #2572 and #2573.

Logs: https://drake-jenkins.csail.mit.edu/job/mac-clang-continuous-matlab/280/consoleFull https://drake-jenkins.csail.mit.edu/job/mac-clang-continuous-matlab/281/consoleFull

Cdash: https://drake-cdash.csail.mit.edu/viewTest.php?onlyfailed&buildid=35597

Output text: 
Invalid MEX-file '/Users/yosemite/workspace/mac-clang-continuous-matlab/drake/systems/plants/inverseKinmex.mexmaci64': dlopen(/Users/yosemite/workspace/mac-clang-continuous-matlab/drake/systems/plants/inverseKinmex.mexmaci64, 6): Symbol not found: __gfortran_transfer_array_write
  Referenced from: /Users/yosemite/workspace/mac-clang-continuous-matlab/drake/pod-build/lib/libdrakeOptimization.dylib
  Expected in: /Applications/MATLAB_R2015b.app/sys/os/maci64/libgfortran.3.dylib
 in /Users/yosemite/workspace/mac-clang-continuous-matlab/drake/pod-build/lib/libdrakeOptimization.dylib

sammy-tri, 2016-06-17:
Possibly related issues:

http://www.mathworks.com/matlabcentral/answers/215947-symbol-not-found-__gfortran_transfer_real_write
https://www.mathworks.com/matlabcentral/answers/17437-dyld_library_path-problem
http://www.mathworks.com/matlabcentral/answers/209708-libgfortran-symbol-not-found
https://projects.coin-or.org/Ipopt/ticket/232

david-german-tri, 2017-04-14:
I'm going to tentatively close this as wontfix.  @sammy-tri, if it's still a pain point for you, please reopen.

closed 2017-04-14 by david-german-tri