RobotLocomotion/drake #2587: Drake Wiki Not Being Updated As Often as Expected

liangfok, 2016-06-18:
https://github.com/RobotLocomotion/drake/pull/2575 was merged about 26 hours ago but the [Drake Wiki page](http://drake.mit.edu/unit_testing_instructions.html#defining-the-build-artifacts-directory) does not reflect the changes yet. Isn't the wiki supposed to update nightly?

liangfok, 2016-06-19:
FTR, the wiki is still not updated this morning.

david-german-tri, 2016-06-20:
Sure enough.  @jamiesnape, any ideas?

jamiesnape, 2016-06-20:
I see the issue. Working on a fix now.

liangfok, 2016-06-21:
FYI. I just noticed that http://drake.mit.edu/from_source.html still does not contain the changes in https://github.com/RobotLocomotion/drake/pull/2555.

jamiesnape, 2016-06-21:
Still working on the fix. I will update this ticket once it is working again.

jamiesnape, 2016-06-22:
@bradking Is the documentation no longer installed into `build/share/doc`?

bradking, 2016-06-22:
See #2558 and RobotLocomotion/drake-ci#3.  If the doc robot is not setting `BUILD_DOCUMENTATION_ALWAYS` then it will not build the docs unless it explicitly builds the `documentation` target.  I'm not aware of any other recent changes to the documentation infrastructure.

The `drake/doc/CMakeLists.txt` file still has:

```cmake
set(DRAKE_DOCUMENTATION_DIR ${CMAKE_INSTALL_PREFIX}/share/doc)
```

jamiesnape, 2016-06-22:
`BUILD_DOCUMENTATION_ALWAYS` and `BUILD_DOCUMENTATION` are both set to be `ON` and I am not seeing anything in `build/share/doc` anymore.

jamiesnape, 2016-06-22:
I think I see the problem. Starting a new build:
https://drake-jenkins.csail.mit.edu/job/linux-gcc-continuous-matlab/360/

jamiesnape, 2016-06-22:
https://github.com/RobotLocomotion/drake/commit/af7323fe00b8769130b56a65cb22d5b74f51ec3a

liangfok, 2016-07-07:
The problem appears to have reappeared. https://github.com/RobotLocomotion/drake/pull/2734 was merged yesterday but its changes have not shown up on http://drake.mit.edu yet.

jamiesnape, 2016-07-07:
Apparently I forgot to toggle the flag on Jenkins. Should be back after build 419 or 420 of linux-gcc-continuous-matlab.

liangfok, 2016-07-08:
LGTM. Closing this issue now.

closed 2016-07-08 by liangfok