RobotLocomotion/drake #2589: Add Model File Name to RigidBody for Debugging Purposes

liangfok, 2016-06-19:
Since multiple URDFs and SDFs can be loaded into the same `RigidBodySystem`, it would be very helpful for debugging purposes to include the name of the SDF or URDF file that resulted in a particular `RigidBody` being created. Such a name could be stored in the `RigidBody` itself. If no file was used to create it, which would be the case if a string specification was used to create the `RigidBody`, a default "[NO FILE NAME]" could be used as the file name.

Also, when running many unit tests, having the file name will be useful when a problem occurs while parsing a model. This is especially true if multiple model files contains models with the same name. For example, currently, both `prius.sdf` and `prius_with_lidar.sdf` has a model called `prius_1`. Of course, one could argue that no two model files should have models with the same name, but such a rule is currently not enforced. Thus, having a file name would be helpful.

liangfok, 2017-04-12:
This might be something to add to #5013. /cc: @amcastro-tri.

closed 2017-04-12 by liangfok

amcastro-tri, 2017-04-12:
This seems to me should belong to the parser infrastructure, not the MBT. Hopefully OSRF's SDF parser will provide an IR that can provide this feature. In the past, RBT was the IR and the MBT.