RobotLocomotion/drake #2598: HybridDrakeSystem Output Frame

manuelli, 2016-06-20:
I am working on some research about control through contact and am using the CompassGait as my toy example. Currently the output  of the HybridDrakeSystem is just the output of the plant for the current mode see [here](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/%40HybridDrakeSystem/HybridDrakeSystem.m#L243). For the Compass Gait example specifically the plants (CompassGaitStancePlant) have only continuous states, and the information about the mode/discrete-states is held in the HybridDrakeSystem. This means that in it's current form it is impossible to get the output of the CompassGait to include the hybrid mode (even though it's part of the state). 

One way to hack around this is to have the actual hybrid mode plants have a discrete state, but I don't think this is a good long term solution. I think that it is natural to be able to include the hybrid mode as part of the output of a HybridDrakeSystem. For example now you can't have the state of the HybridDrakeSystem be the output. When we later want to do feedback with controllers it would make sense to be able to have the hybrid mode included in the output.

A few specific questions for @RussTedrake 

1. Will HybridDrakeSystem continue to live in Matlab or will it be ported over to C++.
2. Is there a working CompassGait example using URDF's rather than hand-written plant functions. I looked in the dev folder but runPassiveURDF threw an error about planar joints not being implemented.
3. What is the preferred way to setup a CompassGait example with a controller where the controller has access to the mode of the system?

RussTedrake, 2016-06-20:
You can make a new class that is derived from HybridDrakeSystem which overloads the output method with the output you need.

Long term it will definitely get to c++, but the minimal hybrid models aren't on the short list yet.

manuelli, 2016-06-21:
Another issue that popped up. I am trying to simulate a CompassGait together with a controller. For the controller I created a direct feedthrough DrakeSystem. I made sure that the input and output frames matched up with those of the CompassGaitPlant. However the feedback(compassGaitPlant, controller) call fails. It seems that the issue is that this feedback call goes bypasses DrakeSystem and attempts the feedback directly in DynamicalSystem. Specifically [here](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/%40HybridDrakeSystem/HybridDrakeSystem.m#L331) in the call to 

`      sys = feedback(SimulinkModel(sys1.getModel()),sys2);
` 

when sys1 gets converted into a SimulinkModel it loses the frame information. Thus although the input/output frames of sys1 and sys2 match up, the I/O frames of `SimulinkModel(sys1.getModel()` and `sys2` don't. This then leads the [calls](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/%40DynamicalSystem/DynamicalSystem.m#L138) of 

`sys2 = sys2.inInputFrame(sys1.getOutputFrame);`
 ` sys2 = sys2.inOutputFrame(sys1.getInputFrame);
` 

to fail. Since I know that the frames match in this case commenting out these two lines fixes the problem.

RussTedrake, 2016-06-21:
you’re right.. this line https://github.com/RobotLocomotion/drake/blob/master/drake/systems/%40HybridDrakeSystem/HybridDrakeSystem.m#L331
is sloppy.

can you try replacing it with 
sys = feedback@DynamicalSystem(sys1,sys2)

manuelli, 2016-06-21:
Ok that didn't throw any errors however it seems like feedback isn't doing the right thing. The CompassGaitPlant is saying it is getting u = 0 even though the controller is always outputting u = 1. If I run the same setup with the plant being PendulumPlant() (which is just a DrakeSystem, instead of a HybridDrakeSystem) then everything works fine. So it seems that something about the wiring of HybridDrakeSystem isn't quite right. We can talk more about what to do with this class in person I guess. For now I am going to hack the controller into the update/dynamics method of HybridDrakeSystem. This effectively means that the controller is part of the plant.

RussTedrake, 2016-09-16:
fyi - @jadecastro and @edrumwri are discussing the c++ api for hybrid systems now.

closed 2016-09-16 by RussTedrake