RobotLocomotion/drake #2603: Consider allowing non-virtual interface pattern in `Interface` classes

sherm1, 2016-06-20:
Per discussion in PR #2593, I propose a small exception to the style guide [restrictions on `Interface` classes](https://google.github.io/styleguide/cppguide.html#Interfaces) to support NVI-style interfaces.

The Google style guide restricts [multiple inheritance](https://google.github.io/styleguide/cppguide.html#Multiple_Inheritance) to a single base class that has an implementation, with the rest being pure interfaces. That is reasonable, but the definition of pure interface is slightly too strict. In the PR mentioned above it has the practical consequence of forcing dynamic_casts where none should be required, when implementing a `Clone()` method. I would like us to be able to implement a covariant `Clone()` like this:
```c++
class MyInterface {
 public:
  /// Doxygen for end users.
  unique_ptr<MyInterface> Clone() const {return DoClone();} // FORBIDDEN
 protected:
  /// Doxygen for developers extending this class.
  MyInterface* DoClone() const = 0;
};

class ConcreteThing : public MyInterface {
 public:
  // This is effectively covariant.
  unique_ptr<ConcreteThing> Clone() const {return DoClone();}
 private:
  // This is a true covariant implementation of the pure virtual.
  ConcreteThing* DoClone() const {return new ConcreteThing(*this);}
};
```
The above permits cloning of a `ConcreteThing` to return a safely unique_ptr-ized `ConcreteThing` without a dynamic cast. But the harmless inline method marked "FORBIDDEN" is not allowed by the standard. Note that the style guide justification is partially "Java programmers are familiar with pure interfaces"; I don't think that is a good reason to lose functionality in C++!

I propose to loosen the rule on what constitutes an `Interface` to permit NVI. The extension would be to (a) permit `protected` pure virtuals, and (b) permit public inline interfaces that invoke those interfaces.

sherm1, 2016-06-20:
(Sorry, forgot `virtual` and `override` above; edited.)

david-german-tri, 2016-06-21:
As the author of #2593, I'm opposed to this change. My `ConcreteThing` does not require a clone method that returns a `unique_ptr<ConcreteThing>`, but if I did, I'd just create that method with a different name.  A style guide exception isn't required when such a simple workaround is available.

I'd like to clarify that the existing style rule is not forcing a `dynamic_cast` in the implementation of a `Clone()` method in #2593.  I think @sherm1 is referring to a `dynamic_cast` in a test case, which is done to verify that the `Clone()` method really is returning a pointer to the leaf type.  I agree that if such casts were to appear in production code, it would be a defect.

sherm1, 2016-06-21:
Well, I don't understand why you would ever want `Thing::Clone()` to return anything but a `Thing`! Losing that ability is one of the [common whines](http://stackoverflow.com/questions/6924754/return-type-covariance-with-smart-pointers) about using smart pointers rather than real covariant raw pointers. And the stackoverflow cure is to wrap covariant raw pointers as above.

However, if you don't want to address this in #2593 I will drop the priority here.

david-german-tri, 2016-09-08:
I think this style request has been overtaken by events in System2 (there are many fewer `Interface`s than there used to be), so I'm going to close it as moot for the time being.

closed 2016-09-08 by david-german-tri