RobotLocomotion/drake #2612: Extending BotVisualizer (drake-visualizer)

wxmerkt, 2016-06-22:
I'm planning on extending the drake-visualizer to support features already present in ``BotVisualizer.m`` (video recording), and beyond that with features prepared for in the ``viewer_command_t`` message definition. Additionally, I'd like to add extend the command enum to support resizing of the view, changing the camera position, and saving screenshots remotely.

I've described more details in https://github.com/RobotLocomotion/director/issues/280

Key questions:
- Are there already plans/work being conducted to extend the BotVisualizer?
- Is there any objection to extending the Matlab implementation of BotVisualizer?
- I'm looking for feedback/comments/input or constraints/limitations you're foreseeing

My plan is:
 1. re-enable the functionality provided by ``playbackMovie(obj, xtraj, filename, options)`` with drake-visualizer
 1. extend BotVisualizer with a ``setCameraPosition(x, y, z)`` method, and extend the control command enum correspondingly
 1. extend BotVisualizer with a ``captureFrame``/``captureImage``/``captureScreenshot`` method, and extend the control command enum correspondingly (saving to default folder, or providing file url in command_data)
 1. extend BotVisualizer with a ``setViewSize(width, height)`` method, and extend the control command enum correspondingly

The intention here is to programmatically create screenshots and movies of the drake-visualizer from within Matlab/C++ e.g. during algorithm development.

liangfok, 2016-06-22:
> Are there already plans/work being conducted to extend the BotVisualizer?

Other than https://github.com/RobotLocomotion/drake/issues/1908, I am not aware of any existing plans. I was told (and recently witnessed) that [OpenHumanoids](https://github.com/openhumanoids) has a much more advanced BotVisualizer / Drake Visualizer system but I haven't studied the code yet.

> Is there any objection to extending the Matlab implementation of BotVisualizer?

No objections on my part, but I believe the long-term goal is to implement most functionality in C++ and to make the Matlab code a thin wrapper around the C++ implementation.

> I'm looking for feedback/comments/input or constraints/limitations you're foreseeing

To get past the rigorous review process, I recommend:

1. Writing a high level design document and having at least one platform reviewer issue a LGTM.
2. Incrementally merging the new features into master via a sequence of small pull requests. In all likelihood, you'll finish the overall implementation, but then break it down into smaller pieces that can be individually and incrementally merged into master.

wxmerkt, 2016-06-22:
Thanks @liangfok! Indeed, ``director`` as part of OpenHumanoids has more features and capabilities than the ``drake-visualizer`` as it's used as a user interface (sensor input, planning and control widgets, segmentation algorithms, different sensor visualisation options etc.). I'm going to make the functionality for remote control via the viewer_command_t message available to both the visualizer as well as the director.

It's not a major change to add the features and it's going to be easy to port the functionality over to C++.

Cool, what do you suggest needs to be described in more detail than above or the director issue, and which reviewer do you suggest?

Thanks! :)

liangfok, 2016-06-22:
Looking at your plan described above, I believe it should be sufficient for a platform reviewer to give an initial go / no go opinion. Today's platform reviewer is @jwnimmer-tri. @jwnimmer-tri, do you see any problems with @wxmerkt's above-mentioned plan? I believe it should be OK since it simply adds an additional feature (video capture and playback) to Bot Visualizer and Drake Visualizer.

jwnimmer-tri, 2016-06-22:
I haven't read the whole thread carefully, but would this answer suffice?
(1) From TRI point of view, we don't care at all about matlab changes.  That's RobotLocomotion's responsibility.
(2) Adding a few new methods to `BotVizualizer.h` does not require pre-emptive design review.

I will also observe that the proposed new APIs and messages to BotVisualizer (to configure the viewer) seem to have nothing to do with being a `DrakeSystem`, so could just as well be added to a new class like `CameraController` or `VisualizerController`, which could send the new messages by itself.

RussTedrake, 2017-02-03:
There is new discussion, since you sent this… including an feature update that landing in director from @rdeits , and a more generic rendering interface for drake coming from @david-german-tri.  

Assigning to david to make the connection.

david-german-tri, 2017-02-06:
> The intention here is to programmatically create screenshots and movies of the drake-visualizer from within Matlab/C++ e.g. during algorithm development.

> a more generic rendering interface for drake coming from @david-german-tri

@wxmerkt 

The interface I'm developing will transmit just the scene graph.  It's perfectly reasonable to control the renderer camera from Drake, but it's not part of my goals.

If I'm understanding RobotLocomotion/director#280 correctly, you plan to send control messages on a separate `DRAKE_VIEWER_COMMAND` channel, with their own LCM type(s).  I think that's great, and our work will neither conflict with nor duplicate yours.  Does that sound right to you?

wxmerkt, 2017-02-07:
@david-german-tri Thanks for the update! Yes, I added some extra enums to the message and corresponding control commands (e.g. https://github.com/RobotLocomotion/director/compare/master...openhumanoids:wxm-extend-drake-visualizer-remote-addressing-280). I'll likely switch to the new viewer and happy to wait and follow how it develops. Happy to close this issue - please keep me posted when the new rendering interface lands :).

RussTedrake, 2017-02-12:
+ @seancurtis-tri .  fyi - it would be great/natural to add recording as an option to the drake visualizer realtime playback method.

SeanCurtis-TRI, 2017-02-13:
@RussTedrake "recording" in what sense?  Additional message to the visualizer or a persisted file?

RussTedrake, 2017-02-13:
Recording as in making a video of the resulting sim.  Director has the ability to screen grab to a series of image files (which can then be a aviconv'd into a movie).

SeanCurtis-TRI, 2017-02-13:
I see.  The idea being, the same time you start a playback, you also send a message indicating that this playback should be captured by the Director.  From the c++ api side, it's simply a flag.  That kind of thing.

david-german-tri, 2017-04-18:
@SeanCurtis-TRI I've reassigned this to you.

SeanCurtis-TRI, 2017-04-18:
@david-german-tri receipt acknowledged.

tri-ltyyu, 2019-03-22:
This is likely specific to RigidBodyPlant. If this is applicable to MultBodyPlant/SceneGraph, please file new issue. Closing for now.

closed 2019-03-22 by tri-ltyyu