RobotLocomotion/drake #2619: Consider chunking for automatic differentiation

tkoolen, 2016-06-22:
I just saw a talk at JuliaCon (held in our building) about ForwardDiff.jl, an automatic differentiation package for Julia. It uses the same idea as Eigen's AutoDiffScalar.

First off, his presentation was excellent and explained the concept very well. It also compares against other differentiation methods, including complex step. See here:
https://github.com/jrevels/ForwardDiffPresentation/blob/master/slides.pdf

More importantly for this issue, the speaker mentioned the use of 'chunking'. If you're taking derivatives w.r.t. many variables, the 'AutoDiffScalar's will be large objects in memory. After a certain number of derivatives, this can cause what he called 'stack thrashing'. I'm not sure if that's the correct term, but I'm interpreting it as e.g. running out of L1 cache in the middle of your algorithm (although I'm not an expert on this subject).

In any case, he said that it can be beneficial to run your algorithm a bunch of times with smaller AutoDiffScalars (10 derivatives at a time or so), than to run your algorithm once with a huge derivative vector.

Using this chunking idea (if it can be made transparent to the user) can also make it easier to find an answer to the question of which finite set of AutoDiffScalar types to support.

sherm1, 2016-06-22:
I like that a lot! We can just instantiate a small assortment of powers-of-2 AutoDiffs, say 1, 2, 4, 8, 16, and never go any bigger.

david-german-tri, 2016-06-22:
:+1: very cool.   Stopping explicit instantiations at 16 Drake-wide seems quite conservative to me, but certainly for some number this approach makes a lot of sense!

tkoolen, 2016-06-23:
Obviously the right chunk size to use depends a lot on the function, but as a point of reference, the speaker hard coded a maximum chunk size of 10 in his library.

liangfok, 2016-06-23:
Does this approach impose any assumptions on the nature of the differential equations? Does it ultimately result in the same solution as the all-at-once approach?

tkoolen, 2016-06-23:
I guess if calling the function has side effects then the results could depend on how many times the function was called. I don't think that's a huge limitation, but it's something to keep in mind.

tkoolen, 2016-09-08:
Now that #3321 is in, there's one more thing I'd like to do. I'd like to extract out the chunking part of `jacobian` into a lower level function that takes a function and a *general* `AutoDiffScalar` matrix argument (not necessarily with derivatives initialized to unit vectors).

This function would essentially translate a call to a function with an `AutoDiffScalar<SomeArbitraryDerivativeType>` matrix argument into a few calls to `AutoDiffScalar<SomeFixedMaximumSizeDerivativeType>`.

This would allow us to get rid of the `AutoDiffScalar<VectorXd>` set of instantiations of the `RigidBodyTree` methods, significantly reducing compilation time, and enabling the computation of gradients for mechanisms with more than 73 states without allocating derivative vectors on the heap. The 73 for the fixed maximum size instantiation should then probably be changed to some other, lower number, to be determined based on some profiling of common use cases.

Ideally, I'd also like to see functions with multiple arguments handled correctly, but that's going to take more effort.

RussTedrake, 2020-01-26:
@sherm -- marking this as part of the team:dynamics queue as it might be important for autodiff performance in the future.  (feel free to close and reopen later if that's most appropriate).  @jwnimmer-tri has done considerable thinking/prototyping of what this might look like in the systems framework, but we've not yet ever made it a priority to finish.

sherm1, 2020-01-30:
Because chunking would likely require visible changes in many places in Drake, I would first like to explore using AutoDiffXd re-implemented to use a pre-allocated pool of identical-sized derivative vectors (since they are the same size during any given computation). That would be another way to eliminate the heap churning and hardware cache utilization should be good if we re-use the most-recently-used vector. I know the perfect person to tackle this!

jwnimmer-tri, 2024-03-20:
Closing as "not planned".  If we decide to do chunking, it will be motivated by a specific issue like "make such-and-such faster", not just a standalone feature.

FYI There are other related discussions in other issues around using compressed storage (like SBO tactics) that would be relevant in case we ever pick this up again.

closed 2024-03-20 by jwnimmer-tri