RobotLocomotion/drake #2623: Add Support for Prescribed Motion Constraints

liangfok, 2016-06-23:
One way to simulate position-controlled robots is to use constraints that achieve a prescribed motion. Such constraints can also be used for kinematic simulations of vehicles. This issue tracks the implementation of these constraints.

@RussTedrake, can you point us to the appropriate reference materials for this?

This might be relevant: http://dl.acm.org/citation.cfm?id=37428

amcastro-tri, 2016-06-23:
How could we gain access to that paper?

liangfok, 2016-06-23:
We'll need to [join ACM](http://www.acm.org/membership/join-acm). I believe it costs $99/yr for access to their digital library.

sherm1, 2016-06-23:
I'm a member and can find that paper if you want it (email me). Prescribed motion is a very simple constraint in internal coordinates since we can just write `q(t)=func(t)` and differentiate twice (func(t) is some analytically known, twice differentiable, function of time, e.g. sin(t)). As a holonomic constraint this is just `g(q,t)=q(t)-func(t)` and we enforce `g(t)=0`. But that is not the best way to do prescribed motion because it unnecessarily turns the ODE into a DAE. Prescribed motion is actually localized inverse dynamics so can be solved without constraints using a mixed forward-inverse recursion. Featherstone calls that "hybrid dynamics" and you can read about it ch 9 of his 2008 book. Abhi Jain calls it "generalized articulated body dynamics"; see ch 16 and algorithm 16.2. Simbody implements prescribed motion that way.

liangfok, 2016-06-23:
It sounds like Featherstone's book is perhaps a better reference. _Looking for a copy now..._

amcastro-tri, 2016-06-23:
Chapter 7.5.2, page 126 of my older copy. It's on top of my desk for those interested.

RussTedrake, 2016-06-23:
Agreed with @sherm1 and we already have them implemented (eg with the loop constraints).

The reason we might want to use an explicit inverse dynamics call instead of prescribed motion is if we want to worry about torque  or speed limits.

jwnimmer-tri, 2016-12-22:
Defer until later.

closed 2016-12-22 by jwnimmer-tri