RobotLocomotion/drake #2624: drake_visualizer on Ubuntu 16.04

vmatos, 2016-06-23:
Hello!
Is anyone running the binary distribution on Ubuntu 16.04?

```
~/libraries/drake-distro/build/bin$ ./drake-visualizer 
./drake-visualizer: error while loading shared libraries: libvtkPythonCore.so.5.8: cannot open shared object file: No such file or directory
```

I can't `sudo apt-get install python-vtk`, because it will try to remove my ROS packages.

I tried to compile from source, but it also stops when it reaches:
```
[ 78%] Performing configure step for 'director'
loading initial cache file /home/vmatos/experiments/drake-distro/externals/director/distro/pods/drake-distro/pod-build/tmp/director-cache-Release.cmake
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.
```

On a side note. Is there any place to discuss topics not related to the development of drake?

liangfok, 2016-06-23:
> I can't sudo apt-get install python-vtk, because it will try to remove my ROS packages.

Which ROS packages are conflicting with `python-vtk`?

> On a side note. Is there any place to discuss topics not related to the development of drake?

None that I'm aware of. Are you looking for a "Drake User's Forum"?

vmatos, 2016-06-23:
Yes! Something like a "Drake User's Forum".

`python-vtk` conflicts with `python-vtk6`, which lots of things depend on it.

```
$ apt-cache depends python-vtk6
python-vtk6
  Depends: libvtk6.2
(...)
  Conflicts: python-vtk
(...)

$ apt-cache rdepends python-vtk6
python-vtk6
Reverse Depends:
  libvtk6-dev
(...)

$ apt-cache rdepends libvtk6-dev
libvtk6-dev
Reverse Depends:
  (...)
  libpcl-dev

$ apt-cache rdepends libpcl-dev
libpcl-dev
Reverse Depends:
  libpcl-conversions-dev
  ros-kinetic-rtabmap
  ros-kinetic-pcl-ros
  ros-kinetic-pcl-conversions
(...)
```

liangfok, 2016-06-23:
Unfortunately, I think this github issue tracker is the closest thing we have right now to a Drake User's Forum. 😢 

Regarding the `python-vtk` vs. `python-vtk6` issue, I believe Drake Visualizer is based on Director, which currently uses VTK 5 (https://github.com/RobotLocomotion/drake/issues/1786#issuecomment-191497215). We are in the process of updating it to use VTK7 (https://github.com/RobotLocomotion/drake/issues/2628). Unfortunately, I'm not sure if that will also conflict with VTK 6 (which is what you're using). 

Do you know if VTK 6 and 7 can peacefully coexist?

vmatos, 2016-06-23:
I don't know how to answer your question.

As far as I understand, only VTK6 is available through ubuntu's repositories.

liangfok, 2016-06-23:
I will be installing ROS + Drake in an Ubuntu 16.04 machine soon. Will let you know if I encounter this problem and if I find a work around.

liangfok, 2016-06-23:
I confirm I'm running into the same problem when I try to install ROS with Drake on Ubuntu 16.04.

```
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist. 
```

liangfok, 2016-06-24:
I just wanted to clarify that I marked this issue as "low priority" because Ubuntu 16.04 + ROS is not on the list of [officially supported platforms](http://drake.mit.edu/developers.html#supported-configurations). We'll resolve it as time permits.

david-german-tri, 2016-09-22:
I think this is resolved in the Ubuntu 16.04 install instructions.  (Still not officially supported, but we're getting there!)  If it still doesn't work for you, feel free to reopen.

http://drake.mit.edu/ubuntu_xenial.html

liangfok, 2016-10-06:
I just installed Drake on a clean Ubuntu 16.04 followed by ROS Kinetic and ran into this issue. I will thus re-open it. I believe the problem is Drake Visualizer requires `libvtk5-dev` to be installed whereas ROS Kinetic requires `libvtk6-dev`.

liangfok, 2016-10-06:
I suspect `libvtk6-dev` is installed by ROS because ROS Kinetic requires PCL 1.7.x, which is provided by package `libpcl-dev`, which depends on `libvtk6-dev`.

http://www.ros.org/reps/rep-0003.html#kinetic-kame-may-2016-may-2021

RussTedrake, 2016-10-06:
fyi @patmarion

patmarion, 2016-10-06:
I think that the best option for right now is to have users of ubuntu 16 and ros Kinetic (who are required to apt-get install vtk6) compile vtk5 from source.  The director superbuild already supports this mode. In the cmake configure, just switch USE_SYSTEM_VTK to OFF (it defaults to ON).  When USE_SYSTEM_VTK is OFF, the director superbuild will clone and compile vtk5.

patmarion, 2016-10-06:
To make this work more smoothly for users, we could submit a PR to director to update its superbuild system: instead of defaulting to USE_SYSTEM_VTK=ON, first check that vtk5 is actually installed, and if it isn't then default USE_SYSTEM_VTK=OFF.

jwnimmer-tri, 2016-12-22:
Mission accomplished.

closed 2016-12-22 by jwnimmer-tri