RobotLocomotion/drake #2625: Previously accessible file ${FILENAME} is now inaccessible

jonathan-owens, 2016-06-23:
On [several MATLAB tests in Windows](https://drake-cdash.csail.mit.edu/viewTest.php?onlyfailed&buildid=39656), the test fails with the following (or similar) error:

```
Previously accessible file "C:\Program Files\MATLAB\R2015b\toolbox\matlab\demos\subsref.m" is now inaccessible.
```

I have only seen this issue on Windows, though I cannot claim that it is exclusive to that environment. A quick Google search pointed to [this](https://www.mathworks.com/matlabcentral/answers/259268-standalone-error-accessible-file-is-now-inaccessible-and-mclfreestacktrace) possible solution.

jonathan-owens, 2016-06-23:
This has happened [some more](https://drake-cdash.csail.mit.edu/viewTest.php?onlyfailed&buildid=39902).

david-german-tri, 2016-10-06:
Windows is no longer supported.

closed 2016-10-06 by david-german-tri