RobotLocomotion/drake #2626: Unable to Compile Drake When LCM Is Disabled

liangfok, 2016-06-23:
# The Error

```
-- Checking for module 'lcm'
--   No package 'lcm' found
-- disabling LCM.  lcm types will not be built.

-- Found Java: /usr/bin/java (found version "1.7.0.101") 
-- Found java, building Java LCM-SPY plugins
-- JAVA_COMPILE: /usr/bin/javac
-- JAVA_ARCHIVE: /usr/bin/jar
-- Found java, building Java LCM-SPY plugins
-- 

----- ERROR: lcm-java not found. Not building bot-spy plugins
```

I believe it's a bug that the build system is trying to build the Java LCM-SPY plugins even though LCM is disabled.

I've only tested this on Ubuntu 14.04.4 LTS and `ninja`.

# Method to Reproduce Error
Here are the commands to replicate the problem:

```
$ cd drake-build
$ env CXX=g++-4.9 CC=gcc-4.9 cmake ../drake-distro -G Ninja -DDISABLE_MATLAB=ON -DWITH_LCM=OFF
```

# Other Remarks
I'm not sure if this issue will be automatically fixed by:
- https://github.com/RobotLocomotion/drake/issues/2486
- https://github.com/RobotLocomotion/drake/issues/2341

jwnimmer-tri, 2016-06-23:
For the record, this is not a supported configuration, so is fixed only on a "best effort" basis.

david-german-tri, 2016-08-09:
It's now supported as part of the "minimal" configuration, and appears to be working.  @liangfok, I'll leave it to you to close this issue if you agree!

liangfok, 2016-08-09:
OK I will test it soon. Thanks for the note.

jhoare, 2016-08-09:
Its possible it will still break, since the steps @liangfok took is a little different from the minimal configuration. e.g. Disabling some other external may be the one that disables building LCM-SPY. Just disabling LCM may still not correctly disable the right things.

jwnimmer-tri, 2016-08-09:
Right.  "Turn off all externals" is supported.  "Default, but without LCM" is still best-effort only.

closed 2016-10-12 by david-german-tri