RobotLocomotion/drake #2627: Update addpath_drake.m To Include Optional Source Code Directory Parameter

liangfok, 2016-06-23:
Currently, `drake/addpath_drake.m` assumes that the source code is in a directory called `drake-distro`. While this is the [documented standard](http://drake.mit.edu/from_source.html), very frequently, developers will maintain many copies of the source code in differently named directories. Thus, it would be nice if an optional input parameter can be added to `drake/addpath_drake.m` that allows a non-standard source code path to be added.

FTR: https://reviewable.io/reviews/robotlocomotion/drake/2599#-KKt2-AMHI7Nih24HyfY

david-german-tri, 2016-06-23:
> Currently, drake/addpath_drake.m assumes that the source code is in a directory called drake-distro. 

I don't think this is true (before or after #2599).  Did you mean to refer to the build tree?

liangfok, 2016-06-23:
Sorry, yes, fixed. Thanks!

closed 2017-02-19 by RussTedrake