RobotLocomotion/drake #2628: Migrate Director to VTK7

david-german-tri, 2016-06-23:
(Split from #1786)

Director currently depends on VTK5, which is obsolete and increasingly difficult to support.  @patmarion has approved switching to VTK7.

liangfok, 2016-06-23:
How are we planning on installing VTK7 on Ubuntu 14.04 and 16.04? From source or via a Personal Package archive (PPA) like [this one](https://launchpad.net/~elvstone/+archive/ubuntu/vtk7)?

david-german-tri, 2016-06-23:
The tentative plan from face-to-face meetings last week was: In the superbuild, check the system for VTK7.  If it's present, use it.  Otherwise, build it from source.  The documentation wouldn't include any install step, but choosing to install gets you a faster superbuild.

liangfok, 2016-10-06:
It might be better to first migrate to VTK6: https://github.com/RobotLocomotion/drake/issues/2624#issuecomment-251859089

patmarion, 2016-10-09:
I compiled and ran drake-visualizer with vtk7 on mac.  It only required a few minor updates, but it's enough that director will not be able to support vtk vtk5 and vtk7, so I will not update director until all users are ready.

I'd like to minimize the number of users that have to compile vtk from source.  If there's a vtk6 package for ubuntu 14, then it might be best to go to vtk6 next.  Homebrew now provides vtk7 only, but it would be possible to host our own vtk6 formula and bottles.

I think that the main jump is between vtk 5 and 6, so it might be possible to support both vtk6 and vtk7.

It might be worth updating to Qt5 as well.  In my recent test with vtk7 on mac, i compiled vtk7 from source using qt4 and python2.

RussTedrake, 2016-10-09:
I agree that using the system vtk is preferrable.  it looks like i have vtk7 (w/ qt4, but could have chosen qt5) on my mac via homebrew.

david-german-tri, 2017-02-23:
Here is an updated plan from conversation with @mwoehlke-kitware, @patmarion, @jamiesnape, @kunimatsu-tri.

----------------

Aside and apart from Drake, the Director superbuild needs to support vtk 6.3.  It's typical for projects to link against both Director and other libraries that were precompiled against system vtk.  One common example is PCL.  Those projects must be able to continue doing so on Ubuntu 16.04, so Director can't move to vtk 7 yet.

Given that, it's probably best for simplicity that we keep Drake and the Drake-built Director on vtk 6.3 as well, even if the APIs are compatible enough that we could theoretically go to vtk 7.  On 16.04, the Drake superbuild can provide vtk 6.3 from the system, or from a pre-built binary package.  On 14.04 and Mac, the Drake superbuild will obviously have to package.  I conclude we might as well package for 16.04 also.  

This gives rise to the following tasks:

- [ ]  Create vtk 6.3 binary packages for both Mac and Linux, host them somewhere, and build enough CMake infrastructure so that the superbuilds can download and link against them.

- [ ]  Add CMake logic to Director that allows it to receive vtk 6.3.

- [ ]  Modify Director and the vtk PCL plugin to use the vtk 6 API, and modify the Director superbuild to provide vtk 6.

- [ ]  Cut OH and Drake over to a new SHA of Director, updating their superbuilds to provide vtk 6.

- [ ]  Clean up any vtk 5 specific CMake logic in Director and the various superbuilds.

- [ ]  Add a Bazel repository rule for the vtk binary package.  According to Pat, vtk has numerous library and header files, but Director doesn't use very many of them, and moreover vtk doesn't have many external dependencies.  So, this might not be as challenging as feared.

patmarion, 2017-02-23:
Note, the vtk 6.3 binary packages should link against system Qt and Python libraries (apt-get, homebrew).  A task that is at least as significant as upgrading the vtk version, is to upgrade to Qt5 and support the various Qt 5.x versions that are system/brew installed.

patmarion, 2017-04-07:
I'd like to make a goal to complete this migration next week.  I'm happy to take on testing and development of the remaining tasks.  @mwoehlke-kitware any work in progress I should start from?

mwoehlke-kitware, 2017-04-07:
I'm out of the office ATM and don't have physical access to my development machine, and the VTK OpenGL2 backend doesn't seem to play nice with Xvfb, so I can't get anything to run. However, before I ran into that, I ran into some Qt5 porting issues in the _Python_ side, which suggests I'm getting fairly deep into the pile of things that could go wrong. (Previously, I had immediate SEGV's due to mixed Qt4/Qt5, so I've at least made it past those, and AFAIK I have everything using Qt5 now.)

Here are the repos/branches I am using:

- https://github.com/mwoehlke-kitware/director@vtk7
- https://gitlab.kitware.com/vtk/vtk.git@v7.1.0
- https://github.com/mwoehlke-kitware/ctkPythonConsole@qt5
- https://github.com/intbots/QtPropertyBrowser@master
- https://github.com/commontk/PythonQt@patched-7

It probably makes sense for you to go ahead and look at my ctkPythonBrowser branch and push it to your repo, but you might not want it on master yet; I don't know how to make a PR that asks for a new branch on the target repo or I'd do that. It's probably safe to go ahead and update PythonQt if you're not already on that branch; I believe that branch will work with either Qt4 or Qt5. QtPropertyBrowser you'll notice is a totally different repo; same idea as the one we use currently, but was created independently (from a newer copy of upstream, I'm pretty sure) and targets Qt5.

The next step on our end I expect is going to be expanding the CMake logic in director to support any of VTK 6.2, 6.3 or 7.1. (If you want to play with the above against VTK 6.x, it shouldn't be hard to back out some of the changes on my vtk7 branch. Mostly it will involve changing the version in the `find_package` call, and tweaking the VTK components.)

patmarion, 2017-04-07:
Looks great, thanks!

jamiesnape, 2017-04-07:
I have someone else coming on board next week to finish the migration.

patmarion, 2017-04-09:
I made some more progress on top of @mwoehlke-kitware's work.  Thanks!

I got things to compile and pass all tests on mac and ubuntu 16.04.  I built with vtk 7.1.1 from source.  On mac I used the latest homebrew qt 5.8 and on ubuntu 16.04 I used system qt 5.5.  My work in progress is here (not cleaned up, subject to change and rebases):

    https://github.com/patmarion/director/tree/port-vtk7-qt5

I updated the director superbuild to pull the updated sha1s for dependencies that needed to be ported to qt5.  You can configure and build the director superbuild with:

```
mkdir build && cd build
cmake -DUSE_SYSTEM_VTK=OFF -DUSE_LCM=ON ../director/distro/superbuild
```

On mac it appears that homebrew qt5 is keg only, so it's not found automatically by cmake, so I also configured with `-DQt5_DIR=/usr/local/Cellar/qt/5.8.0_2/lib/cmake/Qt5`.

I put a binary release for ubuntu using vtk7 and qt5 here:

http://people.csail.mit.edu/patmarion/software/director/releases/director-0.1.0-88-g1da067a-ubuntu-16.04.tar.gz

There are still some important issues to be resolved before this port is ready.  And there are issues with event handling in the render view that I have to investigate.  I am currently using QVTKWidget2 and the VTK OpenGL backend (not OpenGL 2).

jamiesnape, 2017-04-10:
@patmarion Are you going to keep working on it or can the VTK developer that is coming on board finish it up this week?

patmarion, 2017-04-10:
Let's coordinate on slack, would very much like to talk about the remaining details.

stonier, 2017-05-04:
Qt5 problems due to them changing an api for opengl widgets mid-stream that breaks VTK6/7. Kitware is now trying to move it to VTK8.

sankhesh, 2017-05-18:
I have a PR (robotlocomotion/director#508) that allows director superbuild to compile against VTK 8 and Qt4 / Qt5.

When compiling director, one can specify the version of Qt to use by:

```js
cmake -DUSE_SYSTEM_VTK:BOOL=OFF -DDD_QT_VERSION:STRING=5 -DUSE_LCM:BOOL=ON ../director/distro/superbuild
```

stonier, 2017-06-06:
Current plan:

* [ ] finalise PR robotlocomotion/director#508 so that director consumes the new drake/vtk8/qt4-5 combos
* [ ] deal with the very few director-dependent scripts, ensure CI coverage somewhere, #6229.
* [ ] drop director as a dependency of drake

RussTedrake, 2017-06-06:
agreed that drake should not depend on director.  
are you also proposing that director be removed from the drake superbuild?  i think having it build easily on (and link against) a basic drake install is important an important workflow?

stonier, 2017-07-31:
Missed this one. Yes, it will disappear from the superbuild when the CMake disappears so we avoid the build and release awkwardness that arises from the cyclic dependency. The only thing that will remain is the drake visualizer binary. To satisfy the workflow, we do have spartan - we should create some CI tests for that. I do not believe there is any right now.

stonier, 2017-07-31:
Going to close this issue as the Dirrector now does work with VTK 8. The other jobs are related to the Bazel - CMake build shift.

closed 2017-07-31 by stonier