RobotLocomotion/drake #2631: Drake library "drakeRBM" should be called "drakeRigidBodyTree"

liangfok, 2016-06-23:
I suspect the "RBM" stands for "Rigid Body Manipulator". Such a class no longer exists in C++ Drake. The new name is "RigidBodyTree". Thus, the library's name should be updated to reflect the name of the software abstraction.

sherm1, 2016-06-23:
Might have been "rigid body mechanics" but I agree that "rigid body tree" is better.

avalenzu, 2016-07-12:
It was definitely "Rigid Body Manipulator" (though I like the backronym, @sherm1!) :+1: for "drakeRigidBodyTree"

david-german-tri, 2016-07-12:
Minor derailment: it would also be really nice for build parallelism to get the parsers out of this library.  Then the many objects that need to link against RigidBodyTree, but not a parser, won't be blocked on compiling the parsers, which is extremely slow.

This isn't a trivial change right now because RBT itself is rife with convenience calls to parsers, but @liangfok has already marked most of those as deprecated.

liangfok, 2016-07-12:
Agreed. My next step is to further decouple the parsers from the `RigidbodyTree` by having the parsers output `ModelSpec` container objects, which are consumed by a soon-to-be-introduced `RigidBodyTree::addModel()` method. This will allow us to hide the logic that assigns each model a unique ID. We can also pull out the URDF and SDF parsers into their own libraries.

jwnimmer-tri, 2016-10-31:
Perhaps hand the library rename over to @amcastro-tri as part of the #3481 work?

amcastro-tri, 2016-10-31:
I can either do it in #3481 or separately. Either way I like the rename to `drakeRigidBodyTree`.

sammy-tri, 2018-07-11:
There is no longer a library with this name (or a relevant analogue).

closed 2018-07-11 by sammy-tri