RobotLocomotion/drake #2633: 2016-06-23 The mac builders seem to be offline

jwnimmer-tri, 2016-06-23:
The mac builders seem to be offline.

Here's the current Build Queue in jenkins:
```
mac-clang-continuous-matlab-open-source   	
mac-clang-continuous-matlab   	
mac-clang-continuous-release   	
mac-clang-ninja-continuous-debug   	
mac-clang-continuous-debug   	
mac-clang-ninja-continuous-release   	
experimental » clang,mac   	
experimental » clang,mac   	
experimental » clang,mac   	
experimental » clang,mac   	
experimental » clang,mac   
```

jonathan-owens, 2016-06-23:
Taking a look now

jonathan-owens, 2016-06-23:
I think they should be back to building now. I logged into vSphere and looked at the VM's and they were all offline. Not sure why or how that happened, but I started them all back up and things seem to be running again...

I am going to close for now and we'll keep any eye on it.

jwnimmer-tri, 2016-06-24:
No mac builders are up right now, but there are 6 mac builds stuck in the queue, some for as long as ~4 hours.

jonathan-owens, 2016-06-24:
@jwnimmer-tri It looks like it's not just the mac builds, but all of them.

jwnimmer-tri, 2016-06-24:
Yeah.  It got worse.

jonathan-owens, 2016-06-24:
I'm looking into it right now.

liangfok, 2016-06-24:
+1 for reaching out to MacStadium.

david-german-tri, 2016-06-24:
The Mac queue remains stalled.

jonathan-owens, 2016-06-24:
Does anyone know our MacStadium account number?

jonathan-owens, 2016-06-24:
I've contacted MacStadium but I do not have the proper information or permissions to actually do anything. This is what he told me:

```
There is not much assistance that we can offer. Someone who has authorization on the account needs to contact us with details of what assistance is needed for us.
```

As I said above, I haven't been able to get in contact with anyone who could do these things.

jwnimmer-tri, 2016-06-24:
TTBOMK, only Kitware has that information.  Or possibly RussTedrake in a pinch.

jonathan-owens, 2016-06-24:
Just heard from Jamie. We're both having trouble logging into vSphere which manages all the Mac VMs. He's filed a ticket for that.

jonathan-owens, 2016-06-24:
From @jamiesnape:

> We appear to be back. Looks like we were just running too many VMs at once.
> We either need more capacity, less continuous jobs, or live with longer waits in the queue.

We can revisit this on Monday. The infrastructure seems to be all working at the moment.

david-german-tri, 2016-06-28:
Has this recurred? We have five mac builds queued (some for 30+ minutes) and zero workers up.

david-german-tri, 2016-06-29:
Leaving this open pending root cause.

jamiesnape, 2016-06-29:
We have the root cause. Too many builds.

> We either need more capacity, less continuous jobs, or live with longer waits in the queue.

jwnimmer-tri, 2016-06-29:
As best I recall, there was a period of time with no mac builders running, and dozens of mac builds in the queue with the little "waiting for too long" clock next to them.  The problem, to my recollection, was not that there were a handful building but many blocked behind them.

Is the root cause that by asking for too many builds, eventually the system fails into a "nothing works" state?

jamiesnape, 2016-06-29:
Yes.

jwnimmer-tri, 2016-06-29:
So then the open ticket (either this one, or else new feature request(s)) would be to either:
- implement some kind of resource count limit, or dwell time, or something to programmatically prevent overwhelming the mac cloud? or
- add a monitoring system so that this failure mode is caught early, and can be manually worked around? or
- something else?

jamiesnape, 2016-06-29:
Scheduling is entirely handled by Jenkins. I can add a limit on numbers of active nodes now, but it does not work well on that plugin, unfortunately, which one of several reasons why it is not enabled now. The real issue for me is having two continuous builds that take 2-3 hours each.

jhoare, 2016-07-06:
Looks like mac builds are stalled again, and there are not any currently running builds. There are two CI builds that have been waiting for about 14 hours for the mac builds to start/finish. 

https://drake-jenkins.csail.mit.edu/job/experimental/2520/
https://drake-jenkins.csail.mit.edu/job/experimental/2521/

jamiesnape, 2016-07-06:
Now a duplicate of #2747.

closed 2016-07-06 by jamiesnape