RobotLocomotion/drake #2634: Possible Regression of RigidBodyTree::doKinematics()

liangfok, 2016-06-23:
Here is the compile-time error:

```
examples/Cars/CMakeFiles/car_sim_lcm_and_ros.dir/car_sim_lcm_and_ros.cc.o: In function `drake::ros::SensorPublisherOdometry<Drake::RigidBodySystem::StateVector>::output(double const&, Eigen::Matrix<double, 0, 1, 0, 0, 1> const&, Eigen::Matrix<double, -1, 1, 0, -1, 1> const&)':
/home/liang/dev/drake-distro/drake/../drake/ros/ros_sensor_publisher_odometry.h:140: undefined reference to `KinematicsCache<Eigen::Block<Eigen::Matrix<double, -1, 1, 0, -1, 1>, -1, 1, false>::Scalar> RigidBodyTree::doKinematics<Eigen::Block<Eigen::Matrix<double, -1, 1, 0, -1, 1>, -1, 1, false>, Eigen::Block<Eigen::Matrix<double, -1, 1, 0, -1, 1>, -1, 1, false> >(Eigen::MatrixBase<Eigen::Block<Eigen::Matrix<double, -1, 1, 0, -1, 1>, -1, 1, false> > const&, Eigen::MatrixBase<Eigen::Block<Eigen::Matrix<double, -1, 1, 0, -1, 1>, -1, 1, false> > const&, bool) const'

```

liangfok, 2016-06-23:
Unfortunately, adding the following explicit instantiation did not work:

```
template DRAKERBM_EXPORT void RigidBodyTree::doKinematics(
    KinematicsCache<Eigen::Block<Eigen::Matrix<double, -1, 1, 0, -1, 1>, -1, 1,
      false>::Scalar>&, bool) const;
```

Here is the compile-time error:

```
/home/liang/dev/drake-distro/drake/systems/plants/RigidBodyTree.cpp:2379:31: error: duplicate explicit instantiation of ‘void RigidBodyTree::doKinematics(KinematicsCache<Scalar>&, bool) const [with Scalar = double]’ [-fpermissive]
       false>::Scalar>&, bool) const
```

jwnimmer-tri, 2016-06-23:
This looks like code on a branch, not master?  Or at least, I'm not seeing that code anywhere here.

My guess at the problem would be missing `drakeRBM` in the CMake rules for the ROS stuff (or possibly there, but in the wrong dependency order).

liangfok, 2016-06-23:
Yes, it is on this branch: https://github.com/liangfok/drake/tree/feature/golfcart_adapter

Unfortunately, the first thing I checked was to ensure `drakeRBM` was being linked:
https://github.com/liangfok/drake/blob/feature/golfcart_adapter/drake/examples/Cars/CMakeLists.txt#L36

jwnimmer-tri, 2016-06-23:
Double check the linker command line that is failing (to make sure everything that's supposed to be there, is there), and that the copy of `drakeRBM.so` that it's linking against provides the symbols that it should be.

liangfok, 2016-06-23:
Here is the linker's command:

```
/usr/lib/ccache/g++-4.9   -Werror=all -Werror=ignored-qualifiers -DGTEST_DONT_DEFINE_FAIL=1 -DGTEST_DONT_DEFINE_SUCCEED=1 -DGTEST_DONT_DEFINE_TEST=1 -Wno-sign-compare -O2 -g -DNDEBUG   examples/Cars/CMakeFiles/car_sim_lcm_and_ros.dir/car_sim_lcm_and_ros.cc.o  -o bin/car_sim_lcm_and_ros -L/home/liang/dev/drake-build/drake/lib  -L/home/liang/dev/drake-build/install/lib lib/libdrakeCars.so lib/libdrakeRBSystem.so lib/libdrakeLCMSystem.so /opt/ros/indigo/lib/libroscpp.so /usr/lib/x86_64-linux-gnu/libboost_signals.so /usr/lib/x86_64-linux-gnu/libboost_filesystem.so /opt/ros/indigo/lib/librosconsole.so /opt/ros/indigo/lib/librosconsole_log4cxx.so /opt/ros/indigo/lib/librosconsole_backend_interface.so /usr/lib/liblog4cxx.so /usr/lib/x86_64-linux-gnu/libboost_regex.so /opt/ros/indigo/lib/libroscpp_serialization.so /opt/ros/indigo/lib/librostime.so /usr/lib/x86_64-linux-gnu/libboost_date_time.so /opt/ros/indigo/lib/libxmlrpcpp.so /opt/ros/indigo/lib/libcpp_common.so /usr/lib/x86_64-linux-gnu/libboost_system.so /usr/lib/x86_64-linux-gnu/libboost_thread.so -lpthread /usr/lib/x86_64-linux-gnu/libconsole_bridge.so /opt/ros/indigo/lib/libtf.so /opt/ros/indigo/lib/libtf2_ros.so /opt/ros/indigo/lib/libactionlib.so /opt/ros/indigo/lib/libmessage_filters.so /opt/ros/indigo/lib/libroscpp.so /usr/lib/x86_64-linux-gnu/libboost_signals.so /usr/lib/x86_64-linux-gnu/libboost_filesystem.so /opt/ros/indigo/lib/libxmlrpcpp.so /opt/ros/indigo/lib/libtf2.so /opt/ros/indigo/lib/libroscpp_serialization.so /opt/ros/indigo/lib/librosconsole.so /opt/ros/indigo/lib/librosconsole_log4cxx.so /opt/ros/indigo/lib/librosconsole_backend_interface.so /usr/lib/liblog4cxx.so /usr/lib/x86_64-linux-gnu/libboost_regex.so /opt/ros/indigo/lib/librostime.so /usr/lib/x86_64-linux-gnu/libboost_date_time.so /opt/ros/indigo/lib/libcpp_common.so /usr/lib/x86_64-linux-gnu/libboost_system.so /usr/lib/x86_64-linux-gnu/libboost_thread.so -lpthread /usr/lib/x86_64-linux-gnu/libconsole_bridge.so -llcm -lgthread-2.0 -lglib-2.0 -lpthread lib/libdrakeOptimization.so lib/libdrakePolynomial.so -L/usr/lib/gcc/x86_64-linux-gnu/4.8 -L/usr/lib/gcc/x86_64-linux-gnu/4.8/../../../x86_64-linux-gnu -L/usr/lib/gcc/x86_64-linux-gnu/4.8/../../../../lib -L/lib/../lib -L/usr/lib/../lib -L/usr/lib/gcc/x86_64-linux-gnu/4.8/../../.. -lipopt -ldl -lcoinmumps -lcoinlapack -lcoinmetis -lcoinblas -lgfortran -lm -lquadmath -lnlopt -lm -lquadmath -lnlopt -lsnopt_c -lf2c lib/libdrakeRigidBodyConstraint.so lib/libdrakeRBM.so lib/libdrakeCollision.so -L/home/liang/dev/drake-build/install/lib -lBulletSoftBody -lBulletDynamics -lBulletCollision -lLinearMath lib/libdrakeJoints.so lib/libdrakeUtil.so lib/libdrakeXMLUtil.so lib/libtinyxml2.so lib/libdrakeGeometryUtil.so lib/libdrakeShapes.so lib/libspruce.so lib/libdrakeCommon.so /opt/ros/indigo/lib/libroscpp.so /usr/lib/x86_64-linux-gnu/libboost_signals.so /usr/lib/x86_64-linux-gnu/libboost_filesystem.so /opt/ros/indigo/lib/libxmlrpcpp.so /opt/ros/indigo/lib/libtf.so /opt/ros/indigo/lib/libtf2_ros.so /opt/ros/indigo/lib/libactionlib.so /opt/ros/indigo/lib/libmessage_filters.so /opt/ros/indigo/lib/libtf2.so -Wl,-rpath,/home/liang/dev/drake-build/drake/lib:/home/liang/dev/drake-build/install/lib:/opt/ros/indigo/lib
```

I see `libRBM.so` in the command, which I assume is sufficient.

And here's the relevant part of the symbol table within the shared object:

```
$ cd drake-build
$ nm -gC --defined-only drake/lib/libdrakeRBM.so | grep doKinematics
00000000000f6e50 W void RigidBodyTree::doKinematics<double>(KinematicsCache<double>&, bool) const
00000000001082c0 W void RigidBodyTree::doKinematics<Eigen::AutoDiffScalar<Eigen::Matrix<double, -1, 1, 0, 73, 1> > >(KinematicsCache<Eigen::AutoDiffScalar<Eigen::Matrix<double, -1, 1, 0, 73, 1> > >&, bool) const
0000000000126ab0 W void RigidBodyTree::doKinematics<Eigen::AutoDiffScalar<Eigen::Matrix<double, -1, 1, 0, -1, 1> > >(KinematicsCache<Eigen::AutoDiffScalar<Eigen::Matrix<double, -1, 1, 0, -1, 1> > >&, bool) const
00000000000f7c60 W KinematicsCache<Eigen::Map<Eigen::Matrix<double, -1, 1, 0, -1, 1> const, 0, Eigen::Stride<0, 0> >::Scalar> RigidBodyTree::doKinematics<Eigen::Map<Eigen::Matrix<double, -1, 1, 0, -1, 1> const, 0, Eigen::Stride<0, 0> >, Eigen::Map<Eigen::Matrix<double, -1, 1, 0, -1, 1> const, 0, Eigen::Stride<0, 0> > >(Eigen::MatrixBase<Eigen::Map<Eigen::Matrix<double, -1, 1, 0, -1, 1> const, 0, Eigen::Stride<0, 0> > > const&, Eigen::MatrixBase<Eigen::Map<Eigen::Matrix<double, -1, 1, 0, -1, 1> const, 0, Eigen::Stride<0, 0> > > const&, bool) const
00000000000f7a40 W KinematicsCache<Eigen::Map<Eigen::Matrix<double, -1, 1, 0, -1, 1>, 0, Eigen::Stride<0, 0> >::Scalar> RigidBodyTree::doKinematics<Eigen::Map<Eigen::Matrix<double, -1, 1, 0, -1, 1>, 0, Eigen::Stride<0, 0> > >(Eigen::MatrixBase<Eigen::Map<Eigen::Matrix<double, -1, 1, 0, -1, 1>, 0, Eigen::Stride<0, 0> > > const&) const
00000000000f7bd0 W KinematicsCache<Eigen::Map<Eigen::Matrix<double, -1, 1, 0, -1, 1>, 0, Eigen::Stride<0, 0> >::Scalar> RigidBodyTree::doKinematics<Eigen::Map<Eigen::Matrix<double, -1, 1, 0, -1, 1>, 0, Eigen::Stride<0, 0> >, Eigen::Map<Eigen::Matrix<double, -1, 1, 0, -1, 1>, 0, Eigen::Stride<0, 0> > >(Eigen::MatrixBase<Eigen::Map<Eigen::Matrix<double, -1, 1, 0, -1, 1>, 0, Eigen::Stride<0, 0> > > const&, Eigen::MatrixBase<Eigen::Map<Eigen::Matrix<double, -1, 1, 0, -1, 1>, 0, Eigen::Stride<0, 0> > > const&, bool) const
[... cut ...] gen::Matrix<Eigen::AutoDiffScalar<Eigen::Matrix<double, -1, 1, 0, -1, 1> >, -1, 1, 0, -1, 1> > const&, Eigen::MatrixBase<Eigen::Matrix<Eigen::AutoDiffScalar<Eigen::Matrix<double, -1, 1, 0, -1, 1> >, -1, 1, 0, -1, 1> > const&, bool) const
```

The closest template type I can fine in the `.so` file is:

```
Eigen::Matrix<double, -1, 1, 0, -1, 1> const, -1, 1, false>::Scalar
```

This differs from the type my app is looking for:

```
Eigen::Matrix<double, -1, 1, 0, -1, 1>, -1, 1, false>::Scalar
```

david-german-tri, 2016-06-23:
I think Markdown ate some critical pieces of the error message in your original post.

liangfok, 2016-06-23:
I tried adding the following explicit instantiation but got a compile-time error:

```
template DRAKERBM_EXPORT void RigidBodyTree::doKinematics(
    KinematicsCache<Eigen::Block<Eigen::Matrix<double, -1, 1, 0, -1, 1> const,
      -1, 1, false>::Scalar>&, bool) const;
```

Here is the error:

```
/home/liang/dev/drake-distro/drake/systems/plants/RigidBodyTree.cpp:2379:38: error: duplicate explicit instantiation of ‘void RigidBodyTree::doKinematics(KinematicsCache<Scalar>&, bool) const [with Scalar = double]’ [-fpermissive]
       -1, 1, false>::Scalar>&, bool) const;
```

liangfok, 2016-06-23:
@david-german-tri: Thanks, I changed the way it is formatted. PTAL.

david-german-tri, 2016-06-23:
Specifically, I think it's

```
template DRAKERBM_EXPORT KinematicsCache<double> RigidBodyTree::doKinematics(
    Eigen::MatrixBase<Eigen::Block<VectorXd, -1, 1, false>> const&,
    Eigen::MatrixBase<Eigen::Block<VectorXd, -1, 1, false>> const&,
    bool) const;
```

liangfok, 2016-06-23:
Thanks. Testing it now.

liangfok, 2016-06-23:
@david-german-tri: Just to clarify, where does the forward declaration go? In `ros_sensor_publisher_odometry.h`?

david-german-tri, 2016-06-23:
`RigidBodyTree.cpp`.

liangfok, 2016-06-24:
@david-german-tri: Thanks! That fixed the problem. I will issue a PR soon.

jwnimmer-tri, 2016-06-24:
I think a preemptive PR is okay, but it'd also be fine to include it in the PR (branch) that actually needs that variation.

closed 2016-06-27 by amcastro-tri