RobotLocomotion/drake #2640: Drake From-Source Installation Instructions Should be Updated to be Linear

liangfok, 2016-06-24:
It currently involves bouncing between `from_source.rst` to various platform-specific instructions and then back to `from_source.rst`. This makes it unnecessarily difficult to follow. Having a linear sequence of instructions, one for each platform configuration, is more scalable as we spin up support for additional platforms like Ubuntu 16.04.

liangfok, 2016-06-24:
Here's an example for Drake / ROS: https://github.com/liangfok/drake/blob/feature/ros_launch_drake/drake/examples/Cars/README_Drake_ROS_Car_Sim.md

sherm1, 2016-06-24:
Is that example linear only because it is single-platform?

liangfok, 2016-06-24:
Yes, I think there should be a separate linear sequence of instructions for each supported platform and configuration. This will result in greater replication across instruction sequences but I think that tradeoff is OK.

jwnimmer-tri, 2016-06-24:
To be clear: this should definitely **not** be a copy-paste of the existing content into a bunch of separate files.  That would be epically disastrous.

We should write a small amount of tooling to output well-factored linear instructions from separate general and specific pieces -- either conditional guards in some preprocessor-like language on an all-in linear document (my preference), or using several smaller files that are unified as part of the sphinx build process (might be simpler to implement).

jwnimmer-tri, 2016-12-22:
Good enough?  Sure.

closed 2016-12-22 by jwnimmer-tri