RobotLocomotion/drake #2644: Prius example performs all-to-all collision dispatch between all objects in the garage.

amcastro-tri, 2016-06-24:
Drake's current dispatch uses an all-to-all approach (O(N^2)). 
We could probably live with that for now as long as N is small (progress on this is in #2596). In the Prius demo, the car should be checked against the garage (including of course the ground) but it is absolutely unnecessary to check if two pillars are colliding since these are static objects fixed to the world. 
However, these checks are happening right now. To be precise, 4121 checks!

This can be verified by checking the size of the array `id_pairs` in [`BulletModel::closestPointsAllToAll`](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/plants/collision/bullet_model.cc#L689) (called from RBS::dynamics).

This is related to the faulty logic (dead code actually) mentioned in #2481.

amcastro-tri, 2016-06-24:
@liangfok: my fears were confirmed. 
@sherm1: let me share my pain with you.

amcastro-tri, 2016-06-24:
The same failure is observed for the Quadrotor example. The warehouse has 7 collision elements, the quadrotor has 4 collision elements. 53 collision pairs are tested while 20 should suffice (there should be a quadrotor collision clique and a warehouse collision clique).
Cliques were introduced in #2299 but that PR got reverted since caused Gurobi based Atlas' tests to fail.

david-german-tri, 2016-06-24:
Is this a duplicate of #1933?  If so, feel free to close that one.

amcastro-tri, 2016-06-24:
Thanks for the reference @david-german-tri. They are the same issue! Here I am just confirming the fact that the simple Prius example model of the garage has the same issues. Working on a series of fixes now.

amcastro-tri, 2017-02-01:
@SeanCurtis-TRI, this has to do with having `O(N^2)` checks in `DrakeCollision::Model`. I just reassigned to you. Feel free to close or reference in a newer issue.