RobotLocomotion/drake #2647: Possible Problem with Linux-based Build

liangfok, 2016-06-25:
A user encountered the following error when building Drake on a clean install of Ubuntu Linux 14.04.4:

```
~/dev/drake-build$ env CXX=g++-4.9 CC=gcc-4.9 cmake ../drake -G Ninja -DCMAKE_BUILD_TYPE:STRING=Release -DDISABLE_MATLAB=ON
-- The C compiler identification is GNU 4.9.3
-- The CXX compiler identification is GNU 4.9.3
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- CMAKE_INSTALL_PREFIX=/home/ubuntu/dev/drake-build/install
-- Found Git: /usr/bin/git (found version "1.9.1") 
CMake Error at /home/ubuntu/tools/cmake-3.5.2-Linux-x86_64/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find Java (missing: Java_JAR_EXECUTABLE Java_JAVAC_EXECUTABLE
  Java_JAVAH_EXECUTABLE Java_JAVADOC_EXECUTABLE) (found suitable version
  "1.7.0.101", minimum required is "1.6")
Call Stack (most recent call first):
  /home/ubuntu/tools/cmake-3.5.2-Linux-x86_64/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:388 (_FPHSA_FAILURE_MESSAGE)
  /home/ubuntu/tools/cmake-3.5.2-Linux-x86_64/share/cmake-3.5/Modules/FindJava.cmake:255 (find_package_handle_standard_args)
  CMakeLists.txt:64 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/ubuntu/dev/drake-build/CMakeFiles/CMakeOutput.log".
```

The user got around the above error by following the instructions here: http://stackoverflow.com/questions/16263556/installing-java-7-on-ubuntu.

But then the same user encountered a second problem:

```
[3/5] Performing build step for 'swig-matlab'
FAILED: swig-matlab-prefix/src/swig-matlab-stamp/swig-matlab-build 
cd /home/ubuntu/dev/drake-build/externals/swig_matlab/swig-matlab-prefix/src/swig-matlab-build && make && /home/ubuntu/tools/cmake-3.5.2-Linux-x86_64/bin/cmake -E touch /home/ubuntu/dev/drake-build/externals/swig_matlab/swig-matlab-prefix/src/swig-matlab-stamp/swig-matlab-build
make[1]: Entering directory `/home/ubuntu/dev/drake-build/externals/swig_matlab/swig-matlab-prefix/src/swig-matlab-build/Source'
/bin/bash /home/ubuntu/dev/drake/externals/swig_matlab/swig-matlab/Tools/config/ylwrap /home/ubuntu/dev/drake/externals/swig_matlab/swig-matlab/Source/CParse/parser.y y.tab.c CParse/parser.c y.tab.h `echo CParse/parser.c | sed -e s/cc$/hh/ -e s/cpp$/hpp/ -e s/cxx$/hxx/ -e s/c++$/h++/ -e s/c$/h/` y.output CParse/parser.output -- yacc -d 
/home/ubuntu/dev/drake/externals/swig_matlab/swig-matlab/Tools/config/ylwrap: line 176: yacc: command not found
make[1]: *** [CParse/parser.c] Error 127
make[1]: Leaving directory `/home/ubuntu/dev/drake-build/externals/swig_matlab/swig-matlab-prefix/src/swig-matlab-build/Source'
make: *** [source] Error 2
ninja: build stopped: subcommand failed.
```

This ticket is to track my attempt at replicating the above errors and, if successful, resolving them.

liangfok, 2016-06-25:
The second error can be solved by installing:

```
$ sudo apt-get install bison
```

jwnimmer-tri, 2016-06-25:
Both of those errors (jdk, bison) appear to be packages that are already listed in http://drake.mit.edu/ubuntu.html -- perhaps the user did not follow those instructions in detail?

jwnimmer-tri, 2016-07-19:
@liangfok Is this issue still serving any purpose?

liangfok, 2016-07-19:
Nope. The user in question was able to successfully install and use Drake.

closed 2016-07-19 by liangfok