RobotLocomotion/drake #2650: Can't enabled documentation target when `DISABLE_MATLAB` is needed

sherm1, 2016-06-27:
The CMake `BUILD_DOCUMENTATION` variable exists at the `drake` level rather than the superbuild (`drake-distro`) level. `DISABLE_MATLAB` is set at the superbuild level. At least on my Windows machine when I attempt to run CMake at the `drake` level it has forgotten that I disabled Matlab and there is no way to set that variable. That prevents me from enabling the documentation target. If I hack in a `set(DISABLE_MATLAB TRUE)` to the `drake/CMakeLists.txt` I can get this to work.

My preferred fix would be to unconditionally generate the `documentation` target and forget the extra CMake variable, but don't build the documentation by default. Then I can just build the `documentation` target when I want to look at the results. Less good but OK: add a `DISABLE_MATLAB` CMake variable to the `drake` level CMakeLists.txt.

david-german-tri, 2016-06-27:
> My preferred fix would be to unconditionally generate the documentation target and forget the extra CMake variable, but don't build the documentation by default.

Agreed.  Actually, I thought @bradking and I made that change two weeks ago.  I'll try to take a look today.

bradking, 2016-06-27:
@david-german-tri our change was #2558 and we did make it not build by default even when the option is on.  We decided to leave the target optional so that diagnostics (warnings) could be issued when certain pieces of the documentation cannot be built due to missing dependencies.

david-german-tri, 2016-06-27:
Right, thanks.  So @sherm1, let me see if I've got this: You want to build the documentation.  The directions say to run `cmake` inside `drake-build/drake` and set `BUILD_DOCUMENTATION=ON`, which will generate the `documentation` target.  However, `cmake` can't see `DISABLE_MATLAB` from inside `drake-build/drake`, so it generates a bunch of unwanted MATLAB targets too.

I think it's clear we should not recommend running `cmake` inside `drake-build/drake`.  I'll update the instructions to recommend editing `CMakeCache.txt` instead.  

Should we also consider switching the default value of `BUILD_DOCUMENTATION` to `ON`?  (I've been running that way locally since #2558.)  @bradking, can you remind me what warnings we were trying to avoid?  How spammy are they, and who would be affected?

bradking, 2016-06-27:
> `cmake` can't see `DISABLE_MATLAB` from inside `drake-build/drake`

That information could be passed through the external project invocation of cmake.

> I think it's clear we should not recommend running cmake inside drake-build/drake. I'll update the instructions to recommend editing CMakeCache.txt instead. 

I don't see a problem with running it directly.  I do it all the time.  Editing CMakeCache.txt by hand is more risky to recommend IMO.

> remind me what warnings we were trying to avoid?

All the warnings we generate in `drake/doc/CMakeLists.txt` will appear by default if people don't have the associated documentation generators available.

david-german-tri, 2016-06-27:
> That information could be passed through the external project invocation of cmake.

Good point, and actually, it looks to me like that is already happening: https://github.com/RobotLocomotion/drake/blob/master/CMakeLists.txt#L183.  When I set `DISABLE_MATLAB=ON` from the superbuild, it is set in `/drake` as well, and re-running `cmake` in `/drake` does not blow it away.  So perhaps I do not understand @sherm1's original complaint after all.

> All the warnings we generate in drake/doc/CMakeLists.txt will appear by default if people don't have the associated documentation generators available.

Hmm.  If we rephrase the warnings a little bit, that doesn't seem like a big deal.

bradking, 2016-06-27:
> If we rephrase the warnings a little bit, that doesn't seem like a big deal.

Yes, though I think we should still have a `BUILD_DOCUMENTATION` option so it can be turned OFF even if the default becomes ON.

sherm1, 2016-06-27:
> So perhaps I do not understand @sherm1's original complaint after all.

My problem likely resulted from my not using the command line on Windows, where it is somewhat unnatural (a.k.a., not following instructions). I was attempting to run the CMake GUI in the `drake-distro/drake` source directory, where there is no `DISABLE_MATLAB` variable. With the `documentation` target always enabled as in #2655, there will be no need to run CMake except at the initial superbuild, a definite improvement!

PR #2655 (2016-06-27): Enable BUILD_DOCUMENTATION by default.
Downgrade missing documentation tools from WARNING to STATUS, so as not to spam users or CDash.

Fixes #2650.

@bradking for feature review, @sherm1 for platform review

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2655)
closed 2016-06-27 by david-german-tri