RobotLocomotion/drake #2657: MATLAB and Python Bindings Are Triggering Deprecation Warnings

liangfok, 2016-06-27:
Here are the warnings:

```
[196/372] Building CXX object bindings/swig/CMakeFiles/rbtMatlab.dir/__/matlab/rbtree_matlabMATLAB_wrap.cxx.o
bindings/matlab/rbtree_matlabMATLAB_wrap.cxx: In function ‘int _wrap_RigidBodyTree_findLink__SWIG_0(int, mxArray**, int, mxArray**)’:
bindings/matlab/rbtree_matlabMATLAB_wrap.cxx:23351:127: warning: ‘RigidBody* RigidBodyTree::findLink(const string&, const string&, int) const’ is deprecated (declared at /home/liang/dev/drake-distro/drake/../drake/systems/plants/RigidBodyTree.h:585): 
DRAKE DEPRECATED: Please use RigidBodyTree::FindBody instead. [-Wdeprecated-declarations]
       result = (RigidBody *)((RigidBodyTree const *)arg1)->findLink((std::string const &)*arg2,(std::string const &)*arg3,arg4);
                                                                                                                               ^
bindings/matlab/rbtree_matlabMATLAB_wrap.cxx: In function ‘int _wrap_RigidBodyTree_findLink__SWIG_1(int, mxArray**, int, mxArray**)’:
bindings/matlab/rbtree_matlabMATLAB_wrap.cxx:23416:122: warning: ‘RigidBody* RigidBodyTree::findLink(const string&, const string&, int) const’ is deprecated (declared at /home/liang/dev/drake-distro/drake/../drake/systems/plants/RigidBodyTree.h:585): 
DRAKE DEPRECATED: Please use RigidBodyTree::FindBody instead. [-Wdeprecated-declarations]
       result = (RigidBody *)((RigidBodyTree const *)arg1)->findLink((std::string const &)*arg2,(std::string const &)*arg3);
                                                                                                                          ^
bindings/matlab/rbtree_matlabMATLAB_wrap.cxx: In function ‘int _wrap_RigidBodyTree_findLink__SWIG_2(int, mxArray**, int, mxArray**)’:
bindings/matlab/rbtree_matlabMATLAB_wrap.cxx:23468:95: warning: ‘RigidBody* RigidBodyTree::findLink(const string&, const string&, int) const’ is deprecated (declared at /home/liang/dev/drake-distro/drake/../drake/systems/plants/RigidBodyTree.h:585): 
DRAKE DEPRECATED: Please use RigidBodyTree::FindBody instead. [-Wdeprecated-declarations]
       result = (RigidBody *)((RigidBodyTree const *)arg1)->findLink((std::string const &)*arg2);
                                                                                               ^
bindings/matlab/rbtree_matlabMATLAB_wrap.cxx: In function ‘int _wrap_RigidBodyTree_findLinkId__SWIG_0(int, mxArray**, int, mxArray**)’:
bindings/matlab/rbtree_matlabMATLAB_wrap.cxx:23734:94: warning: ‘int RigidBodyTree::findLinkId(const string&, int) const’ is deprecated (declared at /home/liang/dev/drake-distro/drake/../drake/systems/plants/RigidBodyTree.h:615): 
DRAKE DEPRECATED: Pleasse use RigidBodyTree::FindBodyIndex instead. [-Wdeprecated-declarations]
       result = (int)((RigidBodyTree const *)arg1)->findLinkId((std::string const &)*arg2,arg3);
                                                                                              ^
bindings/matlab/rbtree_matlabMATLAB_wrap.cxx: In function ‘int _wrap_RigidBodyTree_findLinkId__SWIG_1(int, mxArray**, int, mxArray**)’:
bindings/matlab/rbtree_matlabMATLAB_wrap.cxx:23781:89: warning: ‘int RigidBodyTree::findLinkId(const string&, int) const’ is deprecated (declared at /home/liang/dev/drake-distro/drake/../drake/systems/plants/RigidBodyTree.h:615): 
DRAKE DEPRECATED: Pleasse use RigidBodyTree::FindBodyIndex instead. [-Wdeprecated-declarations]
       result = (int)((RigidBodyTree const *)arg1)->findLinkId((std::string const &)*arg2);
                                                                                         ^
[223/372] Building CXX object bindings/swig/CMakeFiles/_rbtPython.dir/rbtreePYTHON_wrap.cxx.o
bindings/swig/rbtreePYTHON_wrap.cxx: In function ‘PyObject* _wrap_RigidBodyTree_findLink__SWIG_0(PyObject*, PyObject*)’:
bindings/swig/rbtreePYTHON_wrap.cxx:30188:127: warning: ‘RigidBody* RigidBodyTree::findLink(const string&, const string&, int) const’ is deprecated (declared at /home/liang/dev/drake-distro/drake/../drake/systems/plants/RigidBodyTree.h:585): 
DRAKE DEPRECATED: Please use RigidBodyTree::FindBody instead. [-Wdeprecated-declarations]
       result = (RigidBody *)((RigidBodyTree const *)arg1)->findLink((std::string const &)*arg2,(std::string const &)*arg3,arg4);
                                                                                                                               ^
bindings/swig/rbtreePYTHON_wrap.cxx: In function ‘PyObject* _wrap_RigidBodyTree_findLink__SWIG_1(PyObject*, PyObject*)’:
bindings/swig/rbtreePYTHON_wrap.cxx:30253:122: warning: ‘RigidBody* RigidBodyTree::findLink(const string&, const string&, int) const’ is deprecated (declared at /home/liang/dev/drake-distro/drake/../drake/systems/plants/RigidBodyTree.h:585): 
DRAKE DEPRECATED: Please use RigidBodyTree::FindBody instead. [-Wdeprecated-declarations]
       result = (RigidBody *)((RigidBodyTree const *)arg1)->findLink((std::string const &)*arg2,(std::string const &)*arg3);
                                                                                                                          ^
bindings/swig/rbtreePYTHON_wrap.cxx: In function ‘PyObject* _wrap_RigidBodyTree_findLink__SWIG_2(PyObject*, PyObject*)’:
bindings/swig/rbtreePYTHON_wrap.cxx:30304:95: warning: ‘RigidBody* RigidBodyTree::findLink(const string&, const string&, int) const’ is deprecated (declared at /home/liang/dev/drake-distro/drake/../drake/systems/plants/RigidBodyTree.h:585): 
DRAKE DEPRECATED: Please use RigidBodyTree::FindBody instead. [-Wdeprecated-declarations]
       result = (RigidBody *)((RigidBodyTree const *)arg1)->findLink((std::string const &)*arg2);
                                                                                               ^
bindings/swig/rbtreePYTHON_wrap.cxx: In function ‘PyObject* _wrap_RigidBodyTree_findLinkId__SWIG_0(PyObject*, PyObject*)’:
bindings/swig/rbtreePYTHON_wrap.cxx:30593:94: warning: ‘int RigidBodyTree::findLinkId(const string&, int) const’ is deprecated (declared at /home/liang/dev/drake-distro/drake/../drake/systems/plants/RigidBodyTree.h:615): 
DRAKE DEPRECATED: Pleasse use RigidBodyTree::FindBodyIndex instead. [-Wdeprecated-declarations]
       result = (int)((RigidBodyTree const *)arg1)->findLinkId((std::string const &)*arg2,arg3);
                                                                                              ^
bindings/swig/rbtreePYTHON_wrap.cxx: In function ‘PyObject* _wrap_RigidBodyTree_findLinkId__SWIG_1(PyObject*, PyObject*)’:
bindings/swig/rbtreePYTHON_wrap.cxx:30639:89: warning: ‘int RigidBodyTree::findLinkId(const string&, int) const’ is deprecated (declared at /home/liang/dev/drake-distro/drake/../drake/systems/plants/RigidBodyTree.h:615): 
DRAKE DEPRECATED: Pleasse use RigidBodyTree::FindBodyIndex instead. [-Wdeprecated-declarations]
       result = (int)((RigidBodyTree const *)arg1)->findLinkId((std::string const &)*arg2);
```

Is there any way to continue to include deprecated methods in these bindings, but suppress the warnings shown above?

jwnimmer-tri, 2016-06-27:
You could edit the `drake-distro/drake/bindings/swig/CMakeLists.txt` and disable the warning.

sherm1, 2016-06-27:
Nice to see that macro working as intended! Can these bindings be fixed to avoid the deprecated method? They will eventually break anyway since deprecated methods are not forever.

jwnimmer-tri, 2016-06-27:
Yes, the other alternative would be to edit swig's `*.i` file to provide an, e.g., python findLink that is backed by the C++ `findBody`.  That would be additional work, compared to suppressing the warning.

liangfok, 2016-06-27:
I'll probably just disable the warning for now, and then remove the bindings at the same time the methods are actually removed. Thanks for the advice. A PR is in the works.

closed 2016-06-28 by jwnimmer-tri