RobotLocomotion/drake #2659: Certain Drake Header Files Not Being Installed

liangfok, 2016-06-27:
The following header files are not being installed into `drake-build/install/include/`:

1. `drake/solvers/Constraint.h`
2. `drake/solvers/solution_result.h`
3. `systems/plants/pose_map.h`

This prevents external projects that rely on Drake as an external from including these header files. I discovered this while trying to compile the benchmark in https://github.com/RobotLocomotion/drake/issues/2656.

jhoare, 2016-06-29:
Some others that I ran into are: 

1. ``drake/common/eigen_types.h``
2. ``drake/systems/cascade_system.h``
3. ``drake/systems/feedback_system.h``
4. ``drake/systems/n_ary_state.h``
5. ``drake/systems/n_ary_system.h``
6. ``drake/systems/pd_control_system.h``
7. ``drake/util/drakeAppUtil.h``

I set these to install in my catkin build branch. Should I cherry-pick that change out and submit a Pull Request?

liangfok, 2016-06-29:
Yes please!

jhoare, 2016-06-29:
I also forgot to mention, ``examples/Cars/gen`` header files were installed to ``examples/Cars`` So I fixed that as well. I'll go through the examples and try to fix that everywhere.

PR #2700 (2016-07-05): Marked additional header files for installation.
Fixes #2659 as well as allows the Cars Example to be used by an external project.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2700)
closed 2016-07-05 by jwnimmer-tri