RobotLocomotion/drake #2661: Instantiate Bullet model representation of collision elements at a later stage.

amcastro-tri, 2016-06-27:
In PR #2645 adds the capability to flag collision elements as static. This allows the collision dispatcher to use more efficient data structures when an objects does not move. A typical example is the case of static terrains. 

An issue found in PR #2645 is that at the moment of instantiating collision elements from a SDF/URDF parser there is no enough information to determine if the element is static or not (joints were not read at this stage and therefore no mobility information is available). 

The same problem appears when the user adds to a model read from a file collision elements manually, a very common action used to add terrains in Drake. 
The solution proposed in #2645 is to allow the user to flag these element manually increasing the required level of expertise for the user. 

A flexible solution to this issue is to initialize Bullet (FCL or any other actual representation) objects at a later stage with a call to a method `DrakeCollision::Model::Initialize()`. The model is flagged as `initialized_` on this call and later attempts to modify it by the user will throw an exception.

amcastro-tri, 2017-01-04:
I believe #4554 addresses this? if so, please @SeanCurtis-TRI close this issue once #4554 is merged.

SeanCurtis-TRI, 2017-01-05:
Closed by #4554

closed 2017-01-05 by SeanCurtis-TRI