RobotLocomotion/drake #2662: Replace the usage of integer id's by plain raw pointers for collision elements.

amcastro-tri, 2016-06-27:
The user of `DrakeCollision::Model` gets an id (an integer) as a reference to a collision element added to the model. See [here for instance](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/plants/collision/test/model_test.cc#L132):
```c++
ElementId id3 = model->addElement(element_3);
model->updateElementWorldTransform(id3, T_body3_to_world);
```

The id is assigned by the element itself [at construction (cast to int)](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/plants/collision/Element.cpp#L22).

In the example above the user could do for instance `element_3.set_static()` or (once available) `element_3.set_material(material)` **after** adding the element to the model case in which these calls won't have any effect (and right now they don't warn the user).

This interface is not only provided to the user by is extensively used internally for instance [here](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/plants/collision/bullet_model.cc#L689) where collision elements are passed by id's instead of simply as a raw pointer. The code then needs to find the actual pointer in a map (O(n)) with a call to `Model::readElement`.

amcastro-tri, 2017-02-01:
I wonder how much of this will translate to `MultibodyWorld` and what the solution will be. Again. un-assigning myself in my triage.

liangfok, 2017-02-01:
I've recently encountered one scenario where having indices is better than pointers: cloning. Cloning indices is much easier since they can be directly copied.

jwnimmer-tri, 2017-02-01:
... and if you're lucky, they even still have the same meaning after the clone.

SeanCurtis-TRI, 2017-02-02:
Exactly; it really comes down to the nature of the clone.  Is it a shallow or deep copy?  That aspect of the clone is orthogonal of whether you have pointers or indices stashed inside.

SeanCurtis-TRI, 2017-02-02:
FTR this will be going away with GeometryWorld; bodies will only have indirect access to geometry via "handles" of some sort.

amcastro-tri, 2018-08-14:
Closing this -- it refers to deprecated code that we aren't going to modify now.

closed 2018-08-14 by sherm1