RobotLocomotion/drake #2668: 2016-06-28 Jenkis has 80 builds stalled in the queue

jwnimmer-tri, 2016-06-28:
2016-06-28 Jenkins has 80 builds stalled in the queue
```
windows-msvc-64-nightly-matlab-open-source   	
linux-gcc-nightly-matlab-open-source   	
windows-msvc-ninja-64-nightly-matlab-open-source   	
windows-msvc-ninja-32-nightly-matlab-open-source   	
linux-gcc-nightly-coverage-matlab   	
linux-gcc-nightly-coverage-matlab-open-source   	
windows-msvc-32-nightly-matlab-open-source   	
windows-msvc-ninja-32-nightly-matlab   	
linux-gcc-ninja-nightly-matlab   	
windows-msvc-ninja-64-nightly-matlab   	
windows-msvc-64-nightly-matlab   	
windows-msvc-32-nightly-matlab   	
linux-gcc-nightly-matlab   	
linux-gcc-nightly-release   	
linux-clang-nightly-memcheck-valgrind   	
windows-msvc-64-nightly-release   	
linux-clang-ninja-nightly-release   	
linux-gcc-ninja-nightly-debug   	
windows-msvc-32-nightly-release   	
linux-clang-nightly-release   	
linux-clang-nightly-memcheck-msan   	
linux-clang-ninja-nightly-debug   	
windows-msvc-ninja-32-nightly-debug   	
windows-msvc-64-nightly-debug   	
windows-msvc-32-nightly-debug   	
linux-gcc-nightly-debug   	
linux-clang-nightly-include-what-you-use   	
linux-clang-nightly-memcheck-asan   	
windows-msvc-ninja-64-nightly-debug   	
linux-clang-nightly-coverage   	
windows-msvc-ninja-64-nightly-release   	
windows-msvc-ninja-32-nightly-release   	
linux-clang-nightly-memcheck-tsan   	
linux-clang-nightly-debug   	
linux-clang-nightly-scan-build   	
linux-gcc-ninja-nightly-release   	
linux-cpplint-nightly   	
windows-msvc-ninja-32-continuous-matlab   	
windows-msvc-64-continuous-matlab   	
linux-gcc-ninja-continuous-matlab-open-source   	
linux-gcc-ninja-continuous-matlab   	
linux-gcc-continuous-matlab-open-source   	
windows-msvc-ninja-64-continuous-matlab   	
linux-gcc-continuous-matlab   	
windows-msvc-32-continuous-matlab   	
linux-gcc-continuous-debug   	
linux-gcc-ninja-continuous-release   	
windows-msvc-ninja-32-continuous-release   	
windows-msvc-64-continuous-debug   	
windows-msvc-32-continuous-debug   	
linux-gcc-ninja-continuous-debug   	
windows-msvc-32-continuous-release   	
windows-msvc-ninja-32-continuous-debug   	
linux-clang-continuous-release   	
windows-msvc-ninja-64-continuous-debug   	
windows-msvc-64-continuous-release   	
linux-clang-ninja-continuous-release   	
linux-clang-continuous-debug   	
windows-msvc-ninja-64-continuous-release   	
linux-gcc-continuous-release   	
linux-clang-ninja-continuous-debug   	
linux-cpplint-continuous   	
experimental » cpplint,linux   	
experimental » gcc,linux   	
experimental » msvc-ninja-64,windows   	
experimental » gcc,linux_matlab   	
experimental » clang-ninja,linux_debug   	
experimental » clang,linux   	
experimental » msvc-ninja-32,windows   	
experimental » cpplint,linux   	
experimental » gcc,linux   	
experimental » msvc-ninja-64,windows   	
experimental » gcc,linux_matlab   	
experimental » clang-ninja,linux_debug   	
experimental » clang,linux   	
experimental » msvc-ninja-32,windows   	
experimental » msvc-ninja-64,windows   	
experimental » msvc-ninja-32,windows   	
experimental » msvc-ninja-64,windows   	
experimental » msvc-ninja-32,windows
```

Build executor status is just:
```
 drake-linux-2016-06-07-c4.4xlarge (i-9d1cc20d)  (offline)
 drake-matrix-2016-06-07-m4.xlarge (i-ad73ac3d)
1	Idle			
experimental
	#2332	
experimental
	#2334	
experimental
	#2338	
experimental
	#2339	
 drake-windows-2016-06-07-c4.8xlarge (i-6bd10ffb)  (offline)
```

jwnimmer-tri, 2016-06-28:
/CC @jamiesnape (local expert) @jonathan-owens (prior buildcop)

jamiesnape, 2016-06-28:
Wow. I will look into it now.

jamiesnape, 2016-06-28:
This is a strange one. The UNIX slaves thought they were Windows slaves and vice versa. This happened once before several months ago and I thought maybe someone accidentally changed it, but now I am thinking there is some sort of bug.

david-german-tri, 2016-06-28:
Changed issue description accordingly.  Please keep us updated.

jamiesnape, 2016-06-28:
It is fixed now, but so far I have no idea why it happened. I will keep investigating.

jwnimmer-tri, 2016-07-19:
@jamiesnape Perhaps this can be closed, if there's nothing left to diagnose.

closed 2016-07-19 by jamiesnape