RobotLocomotion/drake #2670: Another Missing doKinematics() Forward Declaration

liangfok, 2016-06-28:
Here is the error:

```
undefined reference to
  KinematicsCache<Eigen::Block<Eigen::Matrix<double, -1, 1, 0, -1, 1>, -1, 1, false>::Scalar>
    RigidBodyTree::doKinematics<Eigen::Block<Eigen::Matrix<double, -1, 1, 0, -1, 1>, -1, 1, false>>(
      Eigen::MatrixBase<Eigen::Block<Eigen::Matrix<double, -1, 1, 0, -1, 1>, -1, 1, false> > const&) const
```

jwnimmer-tri, 2016-06-28:
Assuming this is in a private branch again, I don't think these really merit a github issue.  The feature branch can add the declarations it needs directly in its PRs -- either in an isolated PR, as part of other preparatory work, or perhaps best: concurrently with the code that requires them.

liangfok, 2016-06-28:
OK SGTM.

closed 2016-06-28 by liangfok

liangfok, 2016-06-28:
FTR, this issue was resolved as part of this commit: https://github.com/liangfok/drake/commit/be7944015f2ce91af1e80bf0111f3da83f032147.