RobotLocomotion/drake #2676: Extract SDF and URDF parsing Code Out of RigidBodyTree.h/.cpp

liangfok, 2016-06-28:
This will become especially important as we add support for other formats like XODR.

One potential design is to create a factory for each format. The factory would contain a method like:

```
class ParserURDF {

  // Parses a URDF and adds the model within it to @p rigid_body_tree. If
  // @p rigid_body_tree, @p force_elements, or @p sensor_elements are nullptr,
  // a `std::runtime_error` is thrown.
  static void ParserURDF(const std::string& urdf_text, RigidBodyTree* rigid_body_tree,
    ForceElements* force_elements, SensorElements* sensor_elements);
}
```

liangfok, 2016-06-29:
My new plan is to introduce a new class called `Model` that encapsulates everything about a model (joints, loop joints, frames, links, transmission, actuators, and sensors). The output of the URDF andf SDF parsers will be an array of  `Model` objects.

`RigidBodyTree` and `RigidBodySystem` will then be modified to have a `AddModels(const std::Vector<Model>& models)` method that adds the models to them. This will enable us to hide the way the `model_id` is set within `AddModels(...)`. It will also allow us to provide `RigidBodyTree::get_model_ids()` and `RigidBodyTree::get_num_models()` with `O(1)` time complexity since all bookeeping can be done within `AddModels(...)`.

amcastro-tri, 2016-06-29:
I am not sure I understand the pseudo code above, but what about having a factory method within each object that needs to be read from an URDF/SDF file?
In that way the object itself "knows" how to extract the information to build itself from the information in the file. something like:

```c++
RevoluteJoint* SDFfactory(const XMLElement* node, RigidBodyTree& parent_tree) override {
  ... parse joint params from SDF ...
  RevoluteJoint* joint = RevoluteJoint::CreateAndAddToRBT(parent_tree, ... list of params ...);
  return joint;
```

I am assuming the factory method in `Joint` is pure virtual (with signature `Joint* SDFfactory(const XMLElement*, RigidBodyTree&) = 0`) and notice I am using co-variant return types. Also the user will be forced to implement the factory for a new derived joint since the method is pure virtual in the base `Joint` class.

cc'ing @sherm1, @jwnimmer-tri since we had discussions of this type in the past before.

liangfok, 2016-06-29:
Yes it is definitely possible for each class within a model to know how to parse itself. For this first incremental step towards that, I would like to focus on just the API between the `Model` and the `RigidBodyTree` and the `RigidBodySystem`. For this to exist, I'd like to introduce a `Model` class that essentially serves as a container for transferring information between the parsers and the `RigidBodyTree` and `RigidBodySystem`.

liangfok, 2016-08-09:
I'm going to close this issue because the URDF and SDF parsing code was extracted. See the following files:

* drake-distro/drake/systems/plants/parser_sdf.h
* drake-distro/drake/systems/plants/parser_sdf.cc
* drake-distro/drake/systems/plants/parser_urdf.h
* drake-distro/drake/systems/plants/parser_urdf.cc

We'll create a new issue to track the creation of a `Model` abstraction in the future if we decide it's necessary.

closed 2016-08-09 by liangfok