RobotLocomotion/drake #2680: simulation-termination-test failing on 32 bit Windows

sherm1, 2016-06-28:
Just noticed this on Naveen's unrelated PR #2651; might be a build problem introduced by merge of Liang's PR #2660 that wasn't caught there due to Windows build flakiness at the time. Can you check this @liangfok?

```
13:13:32 52: [==========] Running 3 tests from 1 test case.
13:13:32 52: [----------] Global test environment set-up.
13:13:32 52: [----------] 3 tests from SimulationTerminationTest
13:13:32 52: [ RUN      ] SimulationTerminationTest.TerminationConditionDefault
13:13:32 52: unknown file: error: SEH exception with code 0xc0000005 thrown in SetUp().
13:13:32 52: [  FAILED  ] SimulationTerminationTest.TerminationConditionDefault (0 ms)
13:13:32 52: [ RUN      ] SimulationTerminationTest.TerminationConditionAfterHalfSecond
13:13:32 52: unknown file: error: SEH exception with code 0xc0000005 thrown in SetUp().
13:13:32 52: [  FAILED  ] SimulationTerminationTest.TerminationConditionAfterHalfSecond (0 ms)
13:13:32 52: [ RUN      ] SimulationTerminationTest.TerminationConditionTrue
13:13:32 52: unknown file: error: SEH exception with code 0xc0000005 thrown in SetUp().
13:13:32 52: [  FAILED  ] SimulationTerminationTest.TerminationConditionTrue (1 ms)
13:13:32 52: [----------] 3 tests from SimulationTerminationTest (1 ms total)
13:13:32 52: 
13:13:32 52: [----------] Global test environment tear-down
13:13:32 52: [==========] 3 tests from 1 test case ran. (1 ms total)
13:13:32 52: [  PASSED  ] 0 tests.
13:13:32 52: [  FAILED  ] 3 tests, listed below:
13:13:32 52: [  FAILED  ] SimulationTerminationTest.TerminationConditionDefault
13:13:32 52: [  FAILED  ] SimulationTerminationTest.TerminationConditionAfterHalfSecond
13:13:32 52: [  FAILED  ] SimulationTerminationTest.TerminationConditionTrue
```

sherm1, 2016-06-28:
Oh, I see this was already noticed in post-merge comments in #2660. I'll leave the issue live for now.

jwnimmer-tri, 2016-06-28:
Duplicate of #2679.

jwnimmer-tri, 2016-06-28:
This error wasn't missed because of any Jenkins windows flakiness, it was just pure luck.  You have a 50/50 chance of winning the game with heap alignment.  The original PR #2660 passed Jenkins, but then for some reason something else changed on master, so the heap layout shifted, so thus we are tripping the runtime alignment assertion.  In short, yet another #1854 shotgun-at-foot.

closed 2016-06-28 by jwnimmer-tri