RobotLocomotion/drake #2681: drake-build/install/include is being included with -isystem for some targets (gcc, linux)

david-german-tri, 2016-06-28:
Branched from #2582 

This should never happen.  Why is it happening?  Assigning myself provisionally, but anyone with ideas is welcome to jump in.

david-german-tri, 2016-06-28:
Possible culprit: https://github.com/RobotLocomotion/drake/blob/master/drake/CMakeLists.txt#L224

david-german-tri, 2016-06-28:
Another possible culprit: https://github.com/RobotLocomotion/cmake/blob/master/pods.cmake#L546

david-german-tri, 2016-06-28:
Yes.  `lcmtypes_build` is called here:

https://github.com/RobotLocomotion/drake/blob/master/drake/CMakeLists.txt#L275

In turn, it calls `pods_install_pkg_config_file` for `lcmtypes_drake-java`:

https://github.com/RobotLocomotion/cmake/blob/master/lcmtypes.cmake#L283

That `.pc` file includes all of install/include, and boom.

#2582 will fix by deleting the call to `lcmtypes_build`.

mwoehlke-kitware, 2016-06-28:
Hmm, not obvious how pkg-config includes turn into `-isystem`, but okay :smile:. I guess #2483 also relates, then...

mwoehlke-kitware, 2016-06-28:
> Hmm, not obvious how pkg-config includes turn into -isystem

Bah, don't listen to me :blush:. I *thought* I did a regular `grep -R` at some point, but maybe I only did a `git grep` (and of course the offender is not in the drake repo).

jwnimmer-tri, 2016-07-19:
@david-german-tri Now that #2582 has merged, is this defect still present?

david-german-tri, 2016-11-08:
Actually, we still have a bunch of these.

```
drake-distro/build/drake$ cat build.ninja | grep "build/install/include " | wc -l
149
```

david-german-tri, 2016-12-15:
It's possible this is gone now, will check today.

jamiesnape, 2016-12-15:
Looks like `bot_core_lcmtypes` and `robotlocomotion_lcmtypes` are causing `-isystem` to be added, so not quite there yet.

david-german-tri, 2017-01-26:
Downgrading priority because, once we stop dropping Drake stuff in the install directory, there is less danger of viral `isystem` suppressing Drake warnings.

closed 2018-01-10 by jamiesnape