RobotLocomotion/drake #2689: Move up to C++14

david-german-tri, 2016-06-28:
There has been occasional clamor for C++14 features, most prominently `make_unique`. 

- Clang: The Drake-supported versions on Mac and Linux already implement C++14.
- GCC: The Drake-supported version does not implement C++14.  We would need to upgrade to GCC 5.  
- MSVC: C++14 is not fully implemented in any released version, but significant chunks of it are in the Drake-supported Visual Studio 2015.

Proposal: Let's upgrade to Ubuntu 16.04 first, which has GCC 5 by default, and then throw the C++14 switch.  CI will catch Linux/Mac developers who use features that aren't implemented in VS2015.

jwnimmer-tri, 2016-06-28:
It's not quite clear to me what happens to 14.04 support.  Is it dropped, or retained -- and if retained, is Ubuntu 14.04 + Matlab still retained?

jamiesnape, 2016-06-28:
This is a very non-trivial change for CI. In terms of EC2 support, 16.04 is very, very new and I would be very uncomfortable switching over to it so soon.

david-german-tri, 2016-06-28:
@jwnimmer-tri I would assume we'd have to drop Ubuntu 14.04 (or at least bump it to GCC 5) before enabling C++14.  However, #2691 makes me think that you think that's not a given?

@jamiesnape Fair enough.  I didn't mean to imply undue haste.  Let's talk tomorrow about what would be required to spin up a nightly build on 16.04.

jwnimmer-tri, 2016-06-28:
> @jwnimmer-tri I would assume we'd have to drop Ubuntu 14.04 (or at least bump it to GCC 5) before enabling C++14. However, #2691 makes me think that you think that's not a given?

Yup.  With `g++4.9` on Ubuntu 14.04, the PR passed locally (sans matlab).  We will see what Jenkins says.

FTR, I still do support adding Ubuntu 16.04 as a primary supported platform, but the two issues may be decoupled.

jwnimmer-tri, 2016-06-28:
Closed via #2691.

closed 2016-06-28 by jwnimmer-tri