RobotLocomotion/drake #2693: Drake should provide a way for find_package(Drake) to work

jhoare, 2016-06-29:
Drake should provide a way that a user of Drake should be able to have find_package(Drake) in their CMakeLists.txt and get the correct libraries/headers/etc. 

I believe right now this is mostly accomplished by creating pkg-config (.pc) files, but Drake should create cmake versions in addition to/instead of pkg-config. 

This is probably implied by #2483.

RussTedrake, 2016-09-08:
@david-german-tri  -- assigning to you for your opinion / delegation

david-german-tri, 2016-09-08:
@mwoehlke-kitware and I were talking about this just last week.  I think everyone agrees it's a good idea, but there aren't any particular users in mind.  Reassigned to Matthew and left at priority-medium.

david-german-tri, 2016-09-08:
Whoops, this is a duplicate of #3287, which has more discussion and an actual use case, so I'm closing this one.

closed 2016-09-08 by david-german-tri