RobotLocomotion/drake #2694: submodule repository name change not handled gracefully?

david-german-tri, 2016-06-29:
When merging past #2582, users are reporting errors like the following:

```
[31/146] Performing download step for 'lcm'
FAILED: lcm-prefix/src/lcm-stamp/lcm-download 
cd /home/liang/dev/drake-distro && /usr/bin/git submodule update --init --recursive -- externals/lcm && /home/liang/tools/cmake-3.5.2-Linux-x86_64/bin/cmake -E touch /home/liang/dev/drake-build/lcm-prefix/src/lcm-stamp/lcm-download

fatal: reference is not a tree: 74320f0b58216b0425771aaefbe6f6366797bda6
Unable to checkout '74320f0b58216b0425771aaefbe6f6366797bda6' in submodule path 'externals/lcm'
ninja: build stopped: subcommand failed.
```

@liangfok reports that nuking `drake-distro/.git` fixed it for him.  @jwnimmer-tri speculates that the submodules/cmake integration doesn't cope well with changes in the repository URI, which sounds plausible to me.  I'm going to start poking around, but input from CMake experts would be welcome.

david-german-tri, 2016-06-29:
`git submodule sync` appears to have fixed this failure for me (no other cleanup needed).  I think it should be trivial to integrate into the top-level `CMakeLists.txt`.  Giving it a try now.

bradking, 2016-06-29:
See PR #2695 for a fix.

PR #2695 (2016-06-29): Automatically synchronize submodule URL changes
When `AUTO_UPDATE_EXTERNALS` is `ON` we are responsible for keeping external source trees up to date.  Use `git submodule sync` to honor changes to the submodule URLs.

Fixes #2694

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2695)
commit d6168552 (2016-06-29): cmake: Automatically sync submodule URL changes

When AUTO_UPDATE_EXTERNALS is ON we are responsible for keeping
external source trees up to date.  Use `git submodule sync` to
honor changes to the submodule URLs.

Fixes #2694.
closed 2016-06-29 by david-german-tri