RobotLocomotion/drake #2706: Update Build Instructions to Place Build Directory inside drake-distro/

liangfok, 2016-06-30:
See: https://reviewable.io/reviews/robotlocomotion/drake/2599#-KLVALNyjZtSrJtX46FK

sherm1, 2016-06-30:
Are you saying the build directory *must* be under `drake-distro`? That seems like an unreasonable restriction. Or is that just for Matlab?

liangfok, 2016-06-30:
I don't think the build directory _must_ be under `drake-distro`. We should just document that `drake-distro/build` is the _recommended_ location for the build directory since it appears to be part of an existing workflow.

Maybe we strike a compromise and say something like the following:

> For those who are using MATLAB with Drake, the build directory should be located within `drake-distro`, i.e., `drake-distro/build`. For those not using MATLAB with Drake, the build directory can be placed anywhere. The recommended location is alongside `drake-distro` in a directory called `drake-build`.

Would that be OK?

jwnimmer-tri, 2016-06-30:
There should be exactly 1 recommendation in the docs.  Folks are free to do otherwise at their own risk.

liangfok, 2016-06-30:
OK, how about:

> The recommended location for the build directory is `drake-distro/build`. 

Or, should be a little more verbose and add:

> You may be able to locate the build directory elsewhere, but please do so at your own risk.

?

jwnimmer-tri, 2016-06-30:
I don't care about the wording.  I care that there is only 1 supported directory configuration that I have to test when I change the code.

sherm1, 2016-06-30:
The general case should be an arbitrary build directory location, i.e. disjoint from the source directory. "Arbitrary" also includes in-source for those who want it. But I'm concerned that there is some Matlab-specific dependency on in-source builds, in which case I think it is perfectly reasonable to make that a Matlab-specific instruction.

liangfok, 2016-06-30:
@RussTedrake I think we're going to need more details about the existing workflow and why the build directory needs to be located within `drake-distro`.

RussTedrake, 2016-06-30:
in my view drake-distro is just the container (super-build) for the source directories (drake, externals, etc).  so drake-distro build is out of source.  i also think it's cleaner than naming drake-build alongside drake-distro, and my personal workflow -- which I think many people share -- involves having a number of drake-distro clones checked out.  as an added bonus, it continues to work with the pods search logic that other projects have baked in, so they will continue to find the build directories.

liangfok, 2016-06-30:
Thanks. I see now that it's more than just a MATLAB dependency but also a PODs dependency. I'm testing the ability to place the build artifacts in `drake-distro/build` now.

closed 2016-07-06 by liangfok