RobotLocomotion/drake #2708: ASan/MSan/TSan builds are failing after update to CMake bot_core_lcmtypes

jamiesnape, 2016-06-30:
* https://drake-jenkins.csail.mit.edu/job/linux-clang-nightly-memcheck-asan/144/
* https://drake-jenkins.csail.mit.edu/job/linux-clang-nightly-memcheck-msan/144/
* https://drake-jenkins.csail.mit.edu/job/linux-clang-nightly-memcheck-tsan/126/

mwoehlke-kitware, 2016-06-30:
I think we should turn off asan for lcmgen: https://github.com/lcm-proj/lcm/blob/master/lcmgen/lcmgen.c#L45.

I can't figure out how to do that, though; even removing the `-fsanitize=` from the compile flags doesn't seem sufficient...?

mwoehlke-kitware, 2016-06-30:
> I can't figure out how to do that, though; even removing the -fsanitize= from the compile flags doesn't seem sufficient...?

Arf, missed the one in `CMAKE_EXE_LINKER_FLAGS`...

mwoehlke-kitware, 2016-06-30:
...aaaaaaaand libbot is broken because it *isn't* using the san flags, but LCM (newly) *is*...

jwnimmer-tri, 2016-06-30:
IMO those builds are of extremely low value, and we should disable them instead of burning time on them.

jamiesnape, 2016-06-30:
> IMO those builds are of extremely low value, and we should disable them instead of burning time on them.

I strongly disagree. People should be taking more notice of them.

mwoehlke-kitware, 2016-06-30:
I don't think this can be fixed without overhauling the libbot build system... would it be acceptable to disable libbot for the sanitizer builds?

jamiesnape, 2016-06-30:
I think so.

PR #2711 (2016-07-01): Disable clang sanitization in lcmgen
Update LCM to a branch that disables clang {address,memory} sanitization when building lcm-gen. This is needed as lcm-gen is sloppy about its memory management (reasonably so, as non-systemic leaks in a short lived process aren't really a problem), and so building with sanitization results in a broken executable, which causes problems later when we need to generate LCM bindings.

This also fixes a build error on Ubuntu if Lua is installed. (If LCM finds Lua, it tries to build the Lua module by default, but fails due to the headers on Ubuntu being in a non-s [... cut]
closed 2016-07-01 by ggould-tri