RobotLocomotion/drake #2710: Separate the collision engine interface from the actual implementation

amcastro-tri, 2016-06-30:
Our collision engine is designed to separate the user's interface from the actual implementation, right now Bullet. However the current implementation resorts to a number of tricks or hacks obscuring the implementation, incurring in efficiency and making the addition of other implementations (say FCL) difficult. A summary of other issues greatly caused by this design is documented by #2709.

Another big issue is that since collision elements are just a very light weight representation of an actual implementation through an external library such as Bullet or FCL, there is no placeholder in Drake for additional variables/work-space/bookkeeping needed by any of these libraries when creating new collision elements. This causes series problems when attempting to implement more complex collision geometries such as triangle meshes or height maps. Problem previously found in #1944, #2697.

This issue is created to track progress on a solution to these problems.

amcastro-tri, 2016-06-30:
### Design specifications ###

- Separate interface from implementation. This would allow support for different back ends like Bullet and FCL.
- Provide clean and simple (fast) access to collision elements (to the user) and actual implementations (to the developer). See #2662.
- Separate collision world definition from the actual instantiation. See #2661.
- Implement general purpose collision groups/cliques fully managed by the collision engine. See #1793. Provide compatible API for Matlab. Cliques design proposed, implemented and accepted in #2299 (though reverted due to a Gurobi issue not yet resolved).
- Provide a clean, homogeneous and safe API to the user. Relevant: #2044.
- Implement a *materials library* not only of physical materials but also model approximations. Relevant: #57, #1978.

amcastro-tri, 2016-06-30:
A possible solution based on a PIMPL implementation is posted in [this repository](https://github.com/amcastro-tri/Collision-engine/tree/master).

amcastro-tri, 2016-07-11:
cc'ing @avalenzu since you might be interested in this.

SeanCurtis-TRI, 2018-12-05:
Closing as RBT relic.

closed 2018-12-05 by SeanCurtis-TRI