RobotLocomotion/drake #2718: system2.0: can we make it easier to use the state vector as the output vector?

RussTedrake, 2016-07-01:
this is an incredibly common use case, and it feels like there is something missing if we have to put in all of this boilerplate (twice) for even the spring_mass example:

https://github.com/RobotLocomotion/drake/blob/master/drake/examples/spring_mass/spring_mass_system.h#L46

perhaps we could implement some helper class which inherits from both basic_state_vector and basic_vector?

jwnimmer-tri, 2016-07-01:
I would support in general making the required C++ verbiage less; I had the same concern here.

FWIW, I think the solution for my applications is going to be to code-generate all of the `*Vector` classes anyway (including their LCM support), so generating more boilerplate isn't a huge deal.

sherm1, 2016-07-01:
In general state and output are unrelated. It's common to provide at least some of the states as outputs though so it would make sense to make this convenient. I think we're a ways off from that kind of polish though -- for example, as currently conceived states are not contiguous for a system diagram that has several stateful subsystems, while I believe outputs are contiguous and therefore can't be the same type in general.

Currently the API is close to unusable by mortals IMO. But I think we need to push the functionality further before beautifying -- each layer of functionality so far has caused substantial API changes so it is a moving target.

RussTedrake, 2016-07-01:
ok.  marked as backlog.  are issues the right place to put this type of request (or is the design doc still being used as a point of reference)?

sherm1, 2016-07-01:
I think issues are a good place. TODOs in the can be good too but don't permit much discussion.

david-german-tri, 2016-07-25:
+1 for this feature request.  Early iterations of #2448 did have a single class for both the state and output vectors.  I don't remember why it got dropped.

sherm1, 2016-07-25:
> Early iterations of #2448 did have a single class for both the state and output vectors. I don't remember why it got dropped.

Outputs are more general -- there are multiple output ports and they don't have to be vectors. My preference would be to have *neither* State nor Output classes specialized for particular Systems. A generic OutputPort class seems sufficient to specify the attributes of an output port, including the value type. States come in a very limited assortment, each of which provides a vector value that could be a suitable value source for an OutputPort. That would make it easy for an OutputPort to display a State value. I think we should have a general concept of "value source" (dual to "value listener") to facilitate this. It would also come in handy for defining lists of prerequisite dependencies for invalidation.

david-german-tri, 2016-07-25:
> Outputs are more general -- there are multiple output ports and they don't have to be vectors.

Yes, of course, but there can still be a convenience class that satisfies both `VectorInterface` and `StateVectorInterface`.

sherm1, 2016-07-25:
> Yes, of course, but there can still be a convenience class that satisfies both VectorInterface and StateVectorInterface.

I think a "value source" abstraction might cover more bases. Also, "mode" variables are a kind of state variable that has abstract values; it is likely that we will want to push some of them through output ports for the same reason the vector-valued ones are useful to export.

jwnimmer-tri, 2016-08-15:
Cars needs this, so I'm doing it:
https://github.com/jwnimmer-tri/drake/tree/basic-state-and-output-vector

jwnimmer-tri, 2016-08-16:
I've provided one implementation.  I'll leave the ticket open in case David or Sherm want to pursue this further.

david-german-tri, 2016-08-16:
I like @jwnimmer-tri's implementation a lot, and thinks it meets the original request, so I'm going to close the issue.

closed 2016-08-16 by david-german-tri