RobotLocomotion/drake #2721: Proposed System 2 cached-method naming convention

sherm1, 2016-07-01:
In System 2 and related classes we are likely to have *lots* of methods with these semantics:
- caller wants some returned result calculated with respect to a particular Context, which is supplied as the first argument
- method checks whether result is already up to date in the Context's cache
- (if not, computation is performed and the cache updated)
- return a reference to the up-to-date value in the cache

Russ pointed out in PR #2513 review that calling these methods `GetSomething()` makes them seem too cheap and doesn't adequately convey what's going on. I propose that we pick a different verb, use it consistently, and use it only for this meaning. The one on the table at the moment is `Eval`, as in
```c++
// Naming examples, not necessarily real methods.
const StateVector<double>& EvalTimeDerivatives(context);
const double& EvalPotentialEnergy(context);
const Wrench<double>& EvalJointReactionForce(context, joint_id);
```
Not-yet-landed PR #2513 has examples of this naming scheme in use. Thoughts?

P.S. Sorry abbreviation-haters: we're not spelling this "Evaluate"!

rpoyner-tri, 2016-07-01:
Find? Needs no abbreviation but still somewhat mathy.

liangfok, 2016-07-01:
"Eval" SGTM. I suppose "Compute" is an alternative but it is longer.

sherm1, 2016-07-06:
Thanks, @rpoyner-tri. `Find` is plausible but has a search-for connotation that seems misleading here. PR #2725 is merged now using the `Eval` prefix. I'll close this issue; we can reopen if `Eval` seems bad once we have some mileage on it.

closed 2016-07-06 by sherm1