RobotLocomotion/drake #2722: submodule problem with externals/lcm (possibly Windows-specific)

sherm1, 2016-07-01:
Working on PR #2513, I noticed that my "changed files" list now includes `externals/lcm` and `externals/bot_core_lcmtypes`. I did not change those directories intentionally and I certainly don't want to check the modified ones into master.

I tried a clean superbuild and it appears that at least `drake-distro/externals/lcm` gets modified during the superbuild and then shows as a changed file that need to be committed. That doesn't seem right.

Any suggestions as to what I can do to undamage PR #2513's branch? I don't understand what is going on with those externals directories -- are they supposed to be part of my repo?

sherm1, 2016-07-01:
With the modified externals/lcm in PR #2513 CI builds fail with this message:
```
13:43:20 fatal: reference is not a tree: a0181fc5e86677af9bfbc21622df3d074b612c95
13:43:20 Unable to checkout 'a0181fc5e86677af9bfbc21622df3d074b612c95' in submodule path 'externals/lcm'
13:43:20 make[2]: *** [lcm-prefix/src/lcm-stamp/lcm-download] Error 1
13:43:20 make[2]: Leaving directory `/home/ubuntu/workspace/experimental/7a2abcb0/pod-build'
13:43:20 make[1]: *** [CMakeFiles/lcm.dir/all] Error 2
```
That uuid is the one that got committed to the branch somehow. Help, please!

sherm1, 2016-07-01:
I see this was already reported in #2694 and fixed in #2695 but I'm fully merged up and still having problems. @bradking, can you advise?

liangfok, 2016-07-01:
As a workaround, does it work if you create a completely new clone?

sherm1, 2016-07-01:
Seems fine in a new clone but then I don't have my new commits. If I rebase I think I'll pick up the bad external/lcm again. Not sure what to do -- cherry pick? @mwoehlke-kitware, do you have any advice -- I think I trashed my branch by merging #2582.

liangfok, 2016-07-01:
Have you tried running:
``
$ cd drake-bistro
$ git submodule update --recursive
``
?

Another thing to try is to push your branch to your fork on Github, then do a fresh clone and check out your branch again. I don't think the files under version control are the problem. TTBOMK, it's a problem with stale settings conflicting with new settings.

mwoehlke-kitware, 2016-07-01:
If you're seeing submodule update errors like that, `git submodule sync externals/lcm` (replace the submodule name as needed) may help.

sherm1, 2016-07-01:
Thanks @liangfok and @mwoehlke-kitware. I did run git submodule sync but I don't understand how to undo the damage already done. Directories `external/lcm` and `external/bot_core_lcmtypes` appear to be checked in to the repo now (last via PR #2582). Those somehow got modified in my branch and then I apparently committed them, probably during a merge. Now I have my own personal versions of those directories in my branch (you can see them in the changed files list for PR #2513). I don't want those there but don't know how to get rid of them. Normal git operations don't seem to function on `external/lcm` -- it says the "submodule is dirty". Am I doomed?

liangfok, 2016-07-01:
Maybe you can "revert" to what is on master by deleting your `externals/` directory and copy/pasting it from master?

sherm1, 2016-07-01:
> Maybe you can "revert" to what is on master by deleting your externals/ directory and copy/pasting it from master?

I tried that. Submodules apparently are "special" in some way -- the directories themselves don't actually contain the information that is in the submodule (like the uuid). So copying doesn't clean up.

sherm1, 2016-07-01:
I think I'm just going to make a new branch and copy the modified files into it :-(.

sherm1, 2016-07-01:
I gave up on that branch and replaced it with #2725 which is clean. Thanks for the ideas!

closed 2016-07-01 by sherm1

RussTedrake, 2016-07-02:
fwiw - i saw something like this, too.  @bradking -- this is exactly the type of pain that we dealt with often with submodules before, and why we moved away from them.  please advise how we can make this better for people.

bradking, 2016-07-05:
> please advise how we can make this better for people.

Sherm's branch was based originally on a version from `master` where we didn't have the `git submodule sync` fix in place yet.  That was only a problem because a submodule URL was changed without the sync.  Future topics based on current `master` should not have the problem.  Also changing submodule URLs should be fairly rare.