RobotLocomotion/drake #2723: SNOPT Dependency Fails to Compile: Undefined Architecture: x86_64

liangfok, 2016-07-01:
I did a fresh checkout of Drake's master and tried to compile it with SNOPT enabled. The build failed with the following error:

```
[83/109] Performing build step for 'snopt'
FAILED: snopt-prefix/src/snopt-stamp/snopt-build 
cd /Users/liang/dev/drake3-build/externals/snopt && /usr/local/Cellar/cmake/3.5.0/bin/cmake --build . && /usr/local/Cellar/cmake/3.5.0/bin/cmake -E touch /Users/liang/dev/drake3-build/snopt-prefix/src/snopt-stamp/snopt-build
[1/1] Linking C shared module /Users/liang/dev/drake3-distro/externals/snopt/matlab/snoptcmex.mexmaci64
FAILED: /Users/liang/dev/drake3-distro/externals/snopt/matlab/snoptcmex.mexmaci64 
: && /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc   -O2 -g -DNDEBUG  -bundle -Wl,-headerpad_max_install_names  -o /Users/liang/dev/drake3-distro/externals/snopt/matlab/snoptcmex.mexmaci64 cmex/CMakeFiles/snoptcmex.dir/snoptcmex.c.o cmex/CMakeFiles/snoptcmex.dir/mexUtility.c.o cmex/CMakeFiles/snoptcmex.dir/snoptprobdef.c.o cmex/CMakeFiles/snoptcmex.dir/snoptprint.c.o cmex/CMakeFiles/snoptcmex.dir/sncmexlog.c.o -L/Users/liang/dev/drake3-build/externals/snopt/lib  -L/Users/liang/dev/drake3-build/install/lib lib/libsnopt_c.a lib/libf2c.a -lm -Wl,-rpath,/Users/liang/dev/drake3-build/externals/snopt/lib -Wl,-rpath,/Users/liang/dev/drake3-build/install/lib && :
Undefined symbols for architecture x86_64:
  "_mexCallMATLABWithObject", referenced from:
      _userfg_ in snoptprobdef.c.o
  "_mexErrMsgTxt", referenced from:
      _mexFunction in snoptcmex.c.o
      _sncmxSnopt in snoptcmex.c.o
      _sncmxJac in snoptcmex.c.o
      _sncmxSingleStringArg in snoptcmex.c.o
      _sncmxGetksnoptValue in snoptcmex.c.o
      _sncmxInit in snoptcmex.c.o
      _assertString in mexUtility.c.o
      ...
  "_mexPrintf", referenced from:
      _sncmxabort_ in snoptprint.c.o
      _sncmxwritescreen_ in snoptprint.c.o
  "_mxCalloc", referenced from:
      _sncmxSnopt in snoptcmex.c.o
      _sncmxJac in snoptcmex.c.o
      _sncmxSingleStringArg in snoptcmex.c.o
      _sncmxAlloc in snoptcmex.c.o
      _sncmxReAlloc in snoptcmex.c.o
  "_mxCreateDoubleMatrix_730", referenced from:
      _mexFunction in snoptcmex.c.o
      _sncmxSnopt in snoptcmex.c.o
      _sncmxJac in snoptcmex.c.o
      _sncmxSingleStringArg in snoptcmex.c.o
      _sncmxAllocJac in snoptprobdef.c.o
      _sncmxAllocFuns in snoptprobdef.c.o
      _userfg_ in snoptprobdef.c.o
      ...
  "_mxCreateString", referenced from:
      _sncmxSingleStringArg in snoptcmex.c.o
      _userfg_ in snoptprobdef.c.o
  "_mxDestroyArray", referenced from:
      _sncmxFreeJac in snoptprobdef.c.o
      _sncmxFreeFuns in snoptprobdef.c.o
      _userfg_ in snoptprobdef.c.o
  "_mxFree", referenced from:
      _sncmxSnopt in snoptcmex.c.o
      _sncmxJac in snoptcmex.c.o
      _sncmxSingleStringArg in snoptcmex.c.o
      _sncmxReAlloc in snoptcmex.c.o
      _sncmxFreeWorkSpace in snoptcmex.c.o
  "_mxGetM", referenced from:
      _sncmxSnopt in snoptcmex.c.o
      _sncmxJac in snoptcmex.c.o
      _assertScalar in mexUtility.c.o
      _assertRowDim in mexUtility.c.o
  "_mxGetN", referenced from:
      _sncmxSingleStringArg in snoptcmex.c.o
      _assertScalar in mexUtility.c.o
      _assertColDim in mexUtility.c.o
      _userfg_ in snoptprobdef.c.o
  "_mxGetPr", referenced from:
      _mexFunction in snoptcmex.c.o
      _sncmxSnopt in snoptcmex.c.o
      _sncmxJac in snoptcmex.c.o
      _sncmxSingleStringArg in snoptcmex.c.o
      _sncmxAllocFuns in snoptprobdef.c.o
      _userfg_ in snoptprobdef.c.o
  "_mxGetString_730", referenced from:
      _sncmxSnopt in snoptcmex.c.o
      _sncmxJac in snoptcmex.c.o
      _sncmxSingleStringArg in snoptcmex.c.o
  "_mxIsChar", referenced from:
      _assertString in mexUtility.c.o
  "_mxIsNaN", referenced from:
      _userfg_ in snoptprobdef.c.o
  "_mxIsSparse", referenced from:
      _assertSparse in mexUtility.c.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```
I encountered this on both OS X and Ubuntu.

RussTedrake, 2016-09-08:
it probably couldn't find matlab.  can you reproduce, or close this?

liangfok, 2016-09-08:
I'll try to reproduce it now.

liangfok, 2016-09-08:
I was able to produce this on my OS X machine using the following commands:

```
$ cd drake-distro/build
$ cmake .. -DCMAKE_BUILD_TYPE:STRING=RelWithDebInfo -DDISABLE_MATLAB=FALSE -DWITH_DREAL=FALSE -DWITH_SNOPT=TRUE
$ make -j6 >> ~/Downloads/build_log.txt 2>&1
```
I'm not sure if it's because the build system failed to find MATLAB. Here is the entire build log: https://gist.github.com/liangfok/4dc0ff0cfe776998d47e855d88f5e59e

david-german-tri, 2017-01-03:
Not going to spend cycles on MATLAB.  Closing.

closed 2017-01-03 by david-german-tri