RobotLocomotion/drake #2726: Superbuild fails in Windows Debug

sherm1, 2016-07-01:
I created a fresh clone of Drake and attempted a Superbuild in Visual Studio Debug (Win64), which I have done successfully in the recent past. This time it failed, because lcm was apparently looking for `python27_d.lib` which would be a debug version of python if such a thing existed (is it supposed to?). I speculate that this is due to some otherwise-innocent recent CMakeLists change that caused CMake to add its standard debug suffix here where it wasn't before. Here is the error:
```
10>  Link:
10>    C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\bin\x86_amd64\link.exe /ERRORREPORT:QUEUE /OUT:"C:\Users\Sherm\Documents\GitHub\sherm2\drake-distro-build_d\externals\lcm\lib\python2.7\site-packages\lcm\Debug\_lcm.pyd" /INCREMENTAL /NOLOGO kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib ..\lcm\Debug\lcm.lib C:\Python27\libs\python27.lib /MANIFEST /MANIFESTUAC:"level='asInvoker' uiAccess='false'" /manifest:embed /DEBUG /PDB:"C:/Users/Sherm/Documents/GitHub/sherm2/drake-distro-build_d/externals/lcm/lib/python2.7/site-packages/lcm/Debug/_lcm.pdb" /SUBSYSTEM:CONSOLE /TLBID:1 /DYNAMICBASE /NXCOMPAT /IMPLIB:"C:/Users/Sherm/Documents/GitHub/sherm2/drake-distro-build_d/externals/lcm/lcm-python/Debug/lcm-python.lib" /MACHINE:X64  /machine:x64 /DLL "lcm-python.dir\Debug\module.obj"
10>    "lcm-python.dir\Debug\pyeventlog.obj"
10>    "lcm-python.dir\Debug\pylcm.obj"
10>    "lcm-python.dir\Debug\pylcm_subscription.obj"
10>LINK : fatal error LNK1104: cannot open file 'python27_d.lib' [C:\Users\Sherm\Documents\GitHub\sherm2\drake-distro-build_d\externals\lcm\lcm-python\lcm-python.vcxproj]
10>  Done Building Project "C:\Users\Sherm\Documents\GitHub\sherm2\drake-distro-build_d\externals\lcm\lcm-python\lcm-python.vcxproj" (default targets) -- FAILED.
```

jhoare, 2016-07-06:
@mwoehlke-kitware I assigned you because it sounds like this may be related to the lcm CMakeLists update?

bradking, 2016-07-06:
Python's internal ABI changes between Release and Debug configurations.  Their header has a [#pragma comment](https://msdn.microsoft.com/en-us/library/7f0aews7.aspx) that forces linking to the release library (`python27.lib`) or the debug library (`python27_d.lib`) depending on the configuration.  Of course they don't distribute the debug library in the debug library in the default installer, thus making it almost impossible to use Python in Debug builds without compiling it from source to get the Debug library.

I suspect this was working prior to @mwoehlke-kitware's changes because we didn't used to build lcm as Debug regardless of the main build configuration.

I said "almost impossible" above because there is a hack to work around the problem.  See what VTK does in [vtkPython.h](https://gitlab.kitware.com/vtk/vtk/blob/v7.0.0/Utilities/Python/vtkPython.h#L21-56) to trick the python headers into thinking it is a release build.

mwoehlke-kitware, 2016-07-06:
...and on that note, I don't recall *ever* having a working Windows debug build. (But I might not have ever built on Windows prior to working on the LCM update, so...)

sherm1, 2016-07-15:
@mwoehlke-kitware, @bradking: I'm still stuck with this problem -- I can't get a fully-functional Debug build on Windows without LCM. Can this be fixed?

bradking, 2016-07-19:
Unless LCM it taught to use the same trick I mentioned that VTK uses then it cannot be easily built as Debug on Windows with Python.  Therefore I think the solution here is to have our superbuild always configure LCM to build as Release.  We should still be able to link to LCM libraries from Debug drake libraries.

sherm1, 2016-07-19:
@bradking, thanks -- that seems like a great solution to me if it can work. I don't think it is necessary to have a Debug version of LCM to use while debugging Drake. Mixing Debug and Release builds in windows C++ can be ugly but if LCM has a mostly C-like interface it should be feasible.

PR #2854 (2016-07-20): lcm: Avoid Debug compilation on Windows due to conflict with Python
The Debug configuration of lcm cannot compile on Windows because the Python debug library (`python27_d.lib`) is not generally available.  Always compile the lcm external in its Release configuration on Windows.

Fixes #2726.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/2854)
commit 3d9c7452 (2016-07-20): lcm: Avoid Debug compilation on Windows due to conflict with Python

The Debug configuration of lcm cannot compile on Windows because the
Python debug library (`python27_d.lib`) is not generally available.
Always compile the lcm external in its Release configuration on Windows.

Fixes #2726.
closed 2016-07-20 by sammy-tri