RobotLocomotion/drake #2731: drake-visualizer fails with `Fatal Python error: PyThreadState_Get: no current thread`

RussTedrake, 2016-07-03:
i haven't been able to run drake-visualizer on my mac on master.
```
drake008% ./build/bin/drake-visualizer
Fatal Python error: PyThreadState_Get: no current thread
zsh: abort      ./build/bin/drake-visualizer
```

The error seems likely to be related to python version mismatches:
http://stackoverflow.com/questions/15678153/homebrew-python-on-mac-os-x-10-8-fatal-python-error-pythreadstate-get-no-cu

I did rerun all of the installation instructions to make sure homebrew was up to date, then do a completely clean build of drake and still got the same error.  Anybody else see this?  Any suggestions?

RussTedrake, 2016-07-03:
just realized this was the same as #2095 , which appears to have been closed without resolution.

RussTedrake, 2016-07-03:
i don't see any obvious mismatches in version on my system:
```
drake008% python -V
Python 2.7.12

drake008% pwd
/Users/russt/drake-distro/externals/director/distro/pods/drake-distro/pod-build
drake008% more CMakeCache.txt | grep -i python
PYTHON_CONFIG_EXECUTABLE:FILEPATH=/usr/local/bin/python-config
drake008% /usr/local/bin/python-config --includes
-I/usr/local/Cellar/python/2.7.12/Frameworks/Python.framework/Versions/2.7/include/python2.7 -I/usr/local/Cellar/python/2.7.12/Frameworks/Python.framework/Versions/2.7/include/python2.7
```

rdeits, 2016-07-06:
It's possible that it's not the drake-visualizer binary itself, but one of the compiled libraries that it's loading. 

@RussTedrake can you run the directorPython binary?

RussTedrake, 2016-07-06:
Thanks @rdeits.  Yes - I can run directorPython and get to the console just fine.

RussTedrake, 2016-07-06:
i just trolled through all of the CMakeCache.txt files underneath drake-distro and didn't *see* anything obviously pointing to the wrong python directories...

RussTedrake, 2016-07-06:
i also just tried pulling in https://github.com/RobotLocomotion/director/pull/286 (which looks very relevant), and doing a make clean/make in director/pods/drake-distro, but that didn't resolve it.

hongkai-dai, 2016-07-08:
I just checked out a fresh copy of drake, and installed it from source, on my Ubuntu 14.04 machine. I cannot launch drake-visualizer after building drake. When I run the following command in the terminal
```
./DRAKE_DISTRO_FOLDER/build/install/bin/drake-visualizer
```
I receive the following error
```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/hongkai/drake-distro/build/install/lib/python2.7/dist-packages/director/drakevisualizer.py", line 3, in <module>
    from director import lcmUtils
  File "/home/hongkai/drake-distro/build/install/lib/python2.7/dist-packages/director/lcmUtils.py", line 1, in <module>
    import lcm
  File "/home/hongkai/drake-distro/build/install/lib/python2.7/site-packages/lcm/__init__.py", line 3, in <module>
    from ._lcm import LCM, LCMSubscription
ImportError: dynamic module does not define init function (init_lcm)
```
I can run `directorPython` in the same folder, and call `import lcm` without the error. I notice that after I call `director-visualizer`, it shows that it is using Python 2.7.6, and GCC **4.8.2**, while my gcc version is 4.9.3, as shown in the following screenshot
[image]

liangfok, 2016-07-09:
@hongkai-dai, are you sure the problem you're running into is the same as the one mentioned in this issue's description?

Regardless, I just tested a fresh install of Drake on my Ubuntu 14.04.4 machine and `drake-visualizer` worked. When I ran `directorPython`, it also reports that it was compiled using gcc 4.8.2, as shown below. Thus, I don't believe the gcc version  is the problem.

```
$ cd drake-distro/build
$ ./install/bin/directorPython 
Python 2.7.6 (default, Jun 22 2015, 18:01:27) 
[GCC 4.8.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> 
```

Here are some details of what's on my system.
```
$ cd ~/dev/drake-distro/build/install/lib/python2.7/site-packages/lcm
$ ls -lah _lcm.so
-rw-r--r-- 1 liang liang 88K Jul  8 23:25 _lcm.so
$ nm -gC _lcm.so | grep init_lcm
0000000000002450 T init_lcm
```
 I'm curious how it differs from your system.

rdeits, 2016-07-10:
@RussTedrake if you can run directorPython, then I think you should be able to execute all of the import statements in `drakevisualizer.py`. I suspect that one of those imports will cause the error, which should tell us more about what's going on.

liangfok, 2016-07-10:
I just ran into this problem on my OS X machine as well:

```
$ cd drake-distro-4/build
$ ./install/bin/drake-visualizer
Fatal Python error: PyThreadState_Get: no current thread
Abort trap: 6
```

I can run `directorPython` and found the import line causing the problem:

```
drake-distro-4/build( master)$ ./install/bin/directorPython
Python 2.7.11 (default, Jan 22 2016, 08:29:18)
[GCC 4.2.1 Compatible Apple LLVM 7.0.2 (clang-700.1.81)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import math, os
>>> import numpy as np
>>> from director import lcmUtils
Fatal Python error: PyThreadState_Get: no current thread
Abort trap: 6
```
Thus, something in `drake-distro-4/build/install/lib/python2.7/dist-packages/director/lcmUtils.py` appears to be triggering this problem.

liangfok, 2016-07-10:
Following the instructions [here](http://stackoverflow.com/questions/7807541/is-there-a-way-to-know-by-which-python-version-the-pyc-file-was-compiled) on how to determine which version of Python was used to compile `drake-distro/build/install/lib/python2.7/dist-packages/director/lcmUtils.pyc`, I see that they were both compiled using Python 2.7.11, which has a magic number of `03f30d0a`:

```
$ cd drake-distro/build/install/lib/python2.7/dist-packages/director
$ python
Python 2.7.11 (default, Jan 22 2016, 08:29:18)
[GCC 4.2.1 Compatible Apple LLVM 7.0.2 (clang-700.1.81)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> f = open('lcmUtils.pyc')
>>> magic = f.read(4)
>>> magic.encode('hex')
'03f30d0a'
>>> import imp
>>> imp.get_magic().encode('hex')
'03f30d0a'
>>> quit()
```
I believe this is the _only_ version of Python on my machine:

```
$ which python
/usr/local/bin/python

$ ls -lah /usr/local/bin/python
lrwxr-xr-x  1 liang  admin    34B Mar 23 12:11 /usr/local/bin/python -> ../Cellar/python/2.7.11/bin/python

$ ls -lah /usr/local/Cellar/python/
total 0
drwxr-xr-x   3 liang  admin   102B Mar 23 12:11 .
drwxr-xr-x  70 liang  admin   2.3K Jun 14 13:31 ..
drwxr-xr-x  12 liang  admin   408B Mar 23 12:11 2.7.11
```

rdeits, 2016-07-10:
Thanks, @liangfok that's very helpful! Do you happen to have a version of python3 installed as well? It would be installed in `/usr/local/Cellar/python3`

Both your error and @hongkai-dai 's error are consistent with trying to load a module that was compiled for Python 3 in Python 2 or vice versa. Any chance that might have happened in your case?

liangfok, 2016-07-10:
@rdeits, I checked and don't have a `/usr/local/Cellar/python3` directory:

```
$ ls -lah /usr/local/Cellar | grep python*
drwxr-xr-x   3 liang  admin   102B Mar 23 12:11 python
```

Is there a way to tell what version of Python `directoryUtils` and `drake-visualizer` are using or is expecting?

liangfok, 2016-07-10:
Looking in `drake-distro/build/externals/lcm/CMakeCache.txt`, I see that Python 2.7 is being used to build `lcm`, which is consistent with the above-mentioned magic number of `03f30d0a`.
```
//Path to a file.
PYTHON_INCLUDE_DIR:PATH=/usr/include/python2.7

//Path to a library.
PYTHON_LIBRARY:FILEPATH=/usr/lib/libpython2.7.dylib

//Path to a library.
PYTHON_LIBRARY_DEBUG:FILEPATH=PYTHON_LIBRARY_DEBUG-NOTFOUND

//Dependencies for the target
lcm-python_LIB_DEPENDS:STATIC=general;lcm;general;/usr/lib/libpython2.7.dylib;
```

rdeits, 2016-07-10:
Ah, that might be the issue. LCM found the python library in `/usr/lib/`, which I believe is the python distributed with OSX, not the one in homebrew. You need `PYTHON_LIBRARY` to point to the one in `/usr/local/lib`. That's very likely to cause this kind of issue.

RussTedrake, 2016-07-10:
confirmed.  that's the problem for me, as well.  running cmake in lcm reveals:
```
-- Found PythonLibs: /usr/lib/libpython2.7.dylib (found version "2.7.10")
-- Found PythonInterp: /usr/local/bin/python (found version "2.7.12")
```

this seems to be coming from https://github.com/lcm-proj/lcm/blob/master/CMakeLists.txt#L56 which (seemingly unlike other uses of lcm across drake?) invokes find_package on PythonLibs directly.  

@mwoehlke-kitware this is a blocker.  PTAL

RussTedrake, 2016-07-10:
i take that back, other externals do call find_package on PythonLibs, but I've confirmed that on my machine all other externals find the correct pythonlib.  lcm is the only one that finds `/usr/lib/libpython...`

RussTedrake, 2016-07-10:
aha.  found this in the other relevant cmakelists.txt
```

  if(APPLE)
    find_program(PYTHON_CONFIG_EXECUTABLE python-config)
    if (NOT PYTHON_CONFIG_EXECUTABLE)
      message(SEND_ERROR "python-config executable not found, but python is required.")
    endif()
    # using "python-config --prefix" so that cmake always uses the python that is
    # in the users path, this is a fix for homebrew on Mac:
    # https://github.com/Homebrew/homebrew/issues/25118
    execute_process(COMMAND ${PYTHON_CONFIG_EXECUTABLE} --prefix OUTPUT_VARIABLE python_prefix OUTPUT_STRIP_TRAILING_WHITESPACE)
    set(PYTHON_INCLUDE_DIR ${python_prefix}/include/python2.7)
    set(PYTHON_LIBRARY ${python_prefix}/lib/libpython2.7${CMAKE_SHARED_LIBRARY_SUFFIX})
  endif()
```
I've added that to lcm/CMakeLists.txt, confirmed that it resolved the problem (woohoo!), and will PR to lcm.   thanks @rdeits and @liangfok

liangfok, 2016-07-10:
@rdeits: Thanks! Your are correct, I'm now able to run `drake-visualizer` by forcing `lcm` to use the version of Python that comes with Homebrew rather than OS X.

My current workaround is not pretty though. After doing a complete build, I manually edited `drake-distro/build/externals/lcm/CMakeCache.txt` to contain the following paths (which are to the Homebrew version of Python 2.7):

```
//Path to a program.
PYTHON_EXECUTABLE:FILEPATH=/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/bin/python

//Path to a file.
PYTHON_INCLUDE_DIR:PATH=/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/include/python2.7

//Path to a library.
PYTHON_LIBRARY:FILEPATH=/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/libpython2.7.dylib

//Path to a library.
PYTHON_LIBRARY_DEBUG:FILEPATH=PYTHON_LIBRARY_DEBUG-NOTFOUND

//Dependencies for the target
lcm-python_LIB_DEPENDS:STATIC=general;lcm;general;/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/libpython2.7.dylib;
```

I'm wondering how to make LCM's build system automatically find the Homebrew version of Python 2.7 rather than the OS X version of Python 2.7.

liangfok, 2016-07-10:
Oops! Looks like github comments cross mid flight. I just saw https://github.com/lcm-proj/lcm/pull/82 and will test that soon.

mwoehlke-kitware, 2016-07-11:
>  found this in the other relevant cmakelists.txt [patch; see above]

@bradking, would it make sense for upstream FindPythonLibs to do this if PYTHON_EXECUTABLE is set? At least in some form this seems like logic that should belong in CMake proper...

bradking, 2016-07-11:
@mwoehlke-kitware the use of `python-config` to find python libraries should be independent of `PYTHON_EXECUTABLE`.  Yes, it may have a place upstream.

See [CMake Issue 16142](https://gitlab.kitware.com/cmake/cmake/issues/16142) for a collection of `FindPython{Interp,Libs}` deficiencies that are best addressed by a FindPython revamp and integration into a single module with components.

mwoehlke-kitware, 2016-07-11:
@bradking, I was thinking along the lines of, if you're using the interpreter, pick libraries to match. This would break however if you look for the libraries first. I guess the idea would be to unconditionally (try to) use `python-config` as a hint for both?

bradking, 2016-07-13:
> if you're using the interpreter, pick libraries to match

IIRC FindPythonLibs has some logic for this if FindPythonInterp has run first.

> use python-config as a hint for both?

Yes.

mwoehlke-kitware, 2016-07-13:
> IIRC FindPythonLibs has some logic for this if FindPythonInterp has run first.

It does, but it looks at the path where it found the interpreter, rather than running it and asking it for its install prefix. This works okay on some systems, but the OS X Homebrew install creates a (symlinked) `/usr/local/bin/python` — which is what `FindPythonInterp.cmake` finds — but no corresponding library. The real install prefix (containing the needed library) is `/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7`, which can be found with either `python-config --prefix` or `python -c "import sys; print sys.exec_prefix"`.

RussTedrake, 2016-07-20:
resolved in #2775

closed 2016-07-20 by RussTedrake