RobotLocomotion/drake #2732: Bullet build error: btGjkEpaPenetrationDepthSolver.o

samkhal, 2016-07-03:
I'm getting the following build error trying to build drake. I'm running Ubuntu 14.04, MATLAB R2016a.

```
[ 81%] Linking CXX shared library libBulletCollision.so
CMakeFiles/BulletCollision.dir/NarrowPhaseCollision/btGjkEpaPenetrationDepthSolver.o: file not recognized: File truncated
collect2: error: ld returned 1 exit status
make[9]: *** [src/BulletCollision/libBulletCollision.so.2.83] Error 1
make[8]: *** [src/BulletCollision/CMakeFiles/BulletCollision.dir/all] Error 2
make[7]: *** [all] Error 2
make[6]: *** [bullet3-prefix/src/bullet3-stamp/bullet3-build] Error 2
make[5]: *** [CMakeFiles/bullet3.dir/all] Error 2
make[4]: *** [all] Error 2
make[3]: *** [bullet-prefix/src/bullet-stamp/bullet-build] Error 2
make[2]: *** [CMakeFiles/bullet.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [all] Error 2
```

liangfok, 2016-07-03:
This is a very strange error ("file truncated") that I've never seen before. Does the file `btGjkEpaPenetrationDepthSolver.o` exist on your machine but is corrupt? When happens when you try to view its symbols by executing `nm -A` or `objdump -TC` on it? Is the error reproducible using a fresh checkout of Drake's upstream master (`git clone git@github.com:RobotLocomotion/drake.git drake-distro`)?

Maybe there's another error higher up in the build log where `btGjkEpaPenetrationDepthSolver.o` was being created. Try piping the built output to a file using something like `make 2>&1 | tee -a ~/Downloads/build_log.txt` and seeing if there were other errors before the one you cite.

samkhal, 2016-07-03:
I can't reproduce it with a fresh install; a fresh clone builds fine. Not sure what the cause is; the broken install is at the same git commit.

So I'm happy to just switch to the fresh one, but here's some more info if you're interested:

The file btGjkEpaPenetrationDepthSolver.o exists and is created every build, and I've done `git clean -fdx` between builds. The file shows as 0 bytes, so `nm -A` and `objdump -TC` both give no output on that file.

The error I provided was the first one in the log, but it was also followed by this one:
```
[ 98%] Built target lcm-lua
In file included from /home/sam5/drake-distro/externals/lcm/lcm-python/pylcm_subscription.c:1:0:
/home/sam5/drake-distro/externals/lcm/lcm-python/pylcm_subscription.h:4:20: fatal error: Python.h: No such file or directory
 #include <Python.h>
                    ^
compilation terminated.
/home/sam5/drake-distro/externals/lcm/lcm-python/module.c:1:20: fatal error: Python.h: No such file or directory
 #include <Python.h>
                    ^
compilation terminated.
/home/sam5/drake-distro/externals/lcm/lcm-python/pylcm.c:1:76: fatal error: Python.h: No such file or directory
 #include <Python.h> // include first because it contains pre-processor defs
                                                                            ^
compilation terminated.
/home/sam5/drake-distro/externals/lcm/lcm-python/pyeventlog.c:1:20: fatal error: Python.h: No such file or directory
 #include <Python.h>
                    ^
compilation terminated.
no
checking for suffix of object files... make[6]: *** [lcm-python/CMakeFiles/lcm-python.dir/pylcm_subscription.c.o] Error 1
make[6]: *** Waiting for unfinished jobs....
configure: C compiler options are: -O3 -pipe -DNDEBUG -Wimplicit -Wparentheses -Wsequence-point -Wreturn-type -Wcast-qual -Wall -Wno-unknown-pragmas -Wno-l
ong-long  
o
checking whether we are using the GNU C compiler... make[6]: *** [lcm-python/CMakeFiles/lcm-python.dir/pyeventlog.c.o] Error 1
make[6]: *** [lcm-python/CMakeFiles/lcm-python.dir/module.c.o] Error 1
make[6]: *** [lcm-python/CMakeFiles/lcm-python.dir/pylcm.c.o] Error 1
make[5]: *** [lcm-python/CMakeFiles/lcm-python.dir/all] Error 2
make[4]: *** [all] Error 2
make[3]: *** [lcm-prefix/src/lcm-stamp/lcm-build] Error 2
make[2]: *** [CMakeFiles/lcm.dir/all] Error 2
```

This is odd, since I have no problems with my python install (and the other clean install uses python just fine). If I had to guess, the problem might have something to do with anaconda, which had been on my PATH previously (but not anymore). Before I removed if from my PATH, I was getting build errors relating to anaconda files shadowing system libraries. Maybe that caused some issues that lingered?

liangfok, 2016-07-04:
This remains a mystery to me. If you happen to still have the bad distribution, I'd be curious to know if there are any relevant differences between the various `CMakeCache.txt` files in the bad distribution compared to a good distribution.

david-german-tri, 2016-11-04:
Since this is mysterious and unreproducible, I'm going to close it.

closed 2016-11-04 by david-german-tri