RobotLocomotion/drake #2733: Documentation Needed on Unit Test Dependencies

liangfok, 2016-07-03:
We should update Drake's website within a page listing every unit test and what their dependencies are. That way we won't be left wondering why certain unit tests are disabled.

For context, see: https://github.com/RobotLocomotion/drake/issues/2728#issuecomment-230124852

jwnimmer-tri, 2016-11-01:
It seems like a website with this information would go out-of-date immediately.

Are the `CMakeListst.txt` annotations good enough for this now (to close the issue)?
https://github.com/RobotLocomotion/drake/blob/282d0576437ffae167aa8f37dcfa7ef1e7be5a48/drake/examples/Atlas/CMakeLists.txt#L32

jwnimmer-tri, 2016-12-22:
CMake changes already done are good enough.

closed 2016-12-22 by jwnimmer-tri